betaflight/betaflight #2728: BF 3.2.0 - issues with beeper and vbat

TheAngularity, 2017-03-22:
- `beeper -ON_USB` is not working ... beeps nevertheless with inpluged usb
- default values from vbat are no longer displayed in the bfc 3.1.1
- changed values for vbat in cli are accepted ... but also not displayed
[image]

mikeller, 2017-03-22:
- the beeper could be board specific, if it's the USB detection that's not working. What board is it?
- VBat / IBat settings in the GUI are broken because of the overhaul of the battery code in #2699 / #2701. @hydra will make a PR for corresponding changes in the configurator when they are done in cleanflight. Use the CLI for now;
- in general, can you try to open one issue per problem (in this case beeper / battery)? This makes it a lot easier to keep track of the progress of the issues.

TheAngularity, 2017-03-22:
- if you plug in the usb the fc makes a short beep theme after this comes tree long beeps ... with `beeper -ON_USB` the last three beeps are suppressed ... on 3.1.6 this was still going on but not on 3.2.0
- ok, vbat is only an configurator problem ... then I use the cli for the next time until this is fixed
- and for the next time i would open one issue per problem ;-)

TheAngularity, 2017-03-24:
@mikeller @hydra 
back from flight field with testing 3.2.0 and must say the vbat is not working ... no warning beeps if the battery comes empty ... maybe it could be at the new config ... in any case it has not worked with the old settings

mikeller, 2017-03-26:
@TheAngularity Have you checked the battery related settings in CLI? The GUI configuration for the battery settings is still pending a push of updates from cleanflight, and some of the CLI setting names have changed. I was flying 3.2 today, and battery over telemetry worked just fine. (I didn't fly anything down to battery warning, so I couldn't confirm if the beeper works or not.)

TheAngularity, 2017-03-26:
@mikeller 
on vbat i fly default values only `vbat_scale` was adjusted ... maybe it could be the buzzer ... i can not evaluate the log correctly because the blackbox viewer is not compatible

hydra, 2017-03-26:
I'll double check this when i get a chance. However the logic for the buzzer was not changed.  The code that updates the battery state is used for OSD and Buzzer so if low battery warning for OSD work then they should also work for the buzzer.

TheAngularity, 2017-03-28:
@mikeller @hydra 
both problems solved ... was my bad :( ... i have overlooked that `battery_meter_type` was shortened to `battery_meter` ... vbat and buzzer work now again as usual

closed 2017-03-28 by TheAngularity

husseman, 2017-07-01:
Is there a fix for beeper yet?