betaflight/betaflight #2729: Time from Taranis/openTX for blackbox

Yamaford, 2017-03-22:
Now that we are using Lua script to change PID's with OpenTX, Would it now be possible to send the current time to the FC for use in the Blackbox logs? I understand the FC doesn't have the hardware to "KEEP" time but if we could send the current time say when we connect the lipo could this then be used to create a more useful file name? Blackbox would certainly be more useful if say I experienced a symptom at 3:05 vs trying to remember how many times I connected the lipo If I even remembered to delete logs from the previous days/weeks. If it is possible, I'm sure there would be smarter ways to implement it than what I described but knowing if it is even possible would be a starting point. 

Issues I see would be the FC would need this info before the logger boots if using an external logger. I'm not sure how onboard logging works in this regard.

barthalamu, 2017-03-27:
Great idea! I have so many log files that I don't know which flights they went to!

martinbudden, 2017-05-05:
@mikeller , can you remember who wrote the Taranis Lua scripts for the PID changes?

mikeller, 2017-05-05:
@martinbudden: @raphaelcoeffic did most of the ones that are in https://github.com/betaflight/betaflight-tx-lua-scripts .

martinbudden, 2017-05-05:
@raphaelcoeffic , any chance of you writing a taranis lua script to send the time to the FC? It would just need to be done once, and then the FC could put the time in the BB logs.

mikeller, 2017-05-05:
@martinbudden As a suggestion for an improvement on the original feature request: What about putting the time into the header of the log? This way, it will work for onboard flash logs as well, and the 8.3 filename format for SD card logs will not pose a problem.

raphaelcoeffic, 2017-05-05:
Should be quite easy, indeed! Is there a special request for that? Or should it be tricked over the transport protocol itself?

martinbudden, 2017-05-05:
There's not a special request, but we could add one if needed, or we could reuse one of the existing requests if that is "natural" - is there a suitable existing request you are proposing?

pnu, 2017-07-19:
This would very useful. Another use-case (for correct date/time in FC) would be to enable adding it on OSD. That would help in synchronizing the tuning session DVR footage with bb logs, and other events.

pnu, 2017-07-19:
@mikeller - I think the file naming scheme could remain the same (ie. no issue with 8.3 names), but use the time to initialize the file creation time on filesystem, somewhere around here https://github.com/betaflight/betaflight/blob/master/src/main/io/asyncfatfs/asyncfatfs.c#L2603

mikeller, 2018-02-01:
Implemented by #4289.

closed 2018-02-01 by mikeller