betaflight/betaflight #2731: Feature request: crash detection & prevention

tigert, 2017-03-23:
So this came up on the slack channel the other day, and since I didn't see it filed yet, I wrote a feature request issue so it can be discussed. 

Basically there was a video posted where someone had a midair collision while racing a quad in gates, and the flight controller (claimed to be raceflight) recovered from before dropping to the ground.

I think this could be fairly simple to prototype and useful for freestyle or racing.

I would think the simplest case would be if your quad spins faster than the maximum roll rate in the config, or if the spin happens uncommanded, it is likely that the quad hit something when in the air.

In this case it might be interesting to see whether one could avoid a crash by momentarily enabling a self-leveling mode and adding a small burst of throttle to avoid spinning to the ground

martinbudden, 2017-03-23:
@tigert , I've been meaning to do something like this for a long time. I fly brushed quads indoors a lot and I find that when flying in angle mode it is easy to recover from a crash into a wall, but with rate mode it is not. My idea was to detect the crash and correct the crafts angle by the equivalent of going into angle mode, and then once recovered go back into rate mode.

Sadly I've not had any time to work on this. However I'd done most of the big changes I want to do in betaflight, so perhaps I could find some time for this. I'm not promising anything though.

quiljw01, 2017-03-23:
Thank you for starting this @tigert .... it took me a few days but i was going to submit today until i saw this.  

I think this would be useful for micro quads all the way up to large 'freedom class' quads.  However my specific use will be 5 inch quads for FPV racing.  

**Naming** - I propose calling it "Lava mode"  - Like when you were a kid and you jumped from chair to couch to table because you were pretending the floor is LAVA!!!  

**Description** - Propose some form of ground crash prevention.  Designed to use Accelerometer to detect an impact then activate angle mode automatically for "T" milliseconds.  

Enabled/ armed by either a feature or an aux switch.  But doesn't actually do anything until impact is detected.

What is an impact?    impact = a gyro error more than "X" degrees.  So input rate - actual rate.  
So if you clip a gate, or an obstacle, or mid air collision with another quad.  

When does angle mode turn off?  It should turn off when either one of the 2 circumstances are met.  Either the quad is within "Y" degrees from being level, OR user programmable time "T" has surpassed.  

"T" in milliseconds is a timeout that should be in place to prevent too much damage in the case where you're missing a prop and there is no chance for recovery.   Maybe a good value for this is between 50ms and 500ms.  Testing would be required but this can be done by the pilot, not the programmers.  

Rumors say this feature already exists in a different firmware but insufficient evidence exists to prove it.  

#2731

tigert, 2017-03-23:
One thing to be careful with is landing, obviously since upon touchdown you will get a similar situation. 

You could of course have this "lava mode" activation via aux switch so you turn it off before landing, but I would rather avoid long mental "checklists" if it can be avoided.

mikeller, 2017-03-24:
@tigert: Possible mitigation to the 'landing' scenario would be to disable the feature when below a certain throttle value.

quiljw01, 2017-03-24:
i think there won't be an issue on landing any more than when trying to land in AIR Mode.   I think we should NOT disable the feature below a certain throttle value.  Many times when you crash, it is common/ natural to instinctively lower the throttle, which would then defeat the purpose of the feature in the first place.

bennymack, 2017-03-24:
With regards to activating this feature with a switch, I believe there are no more available "boxes" as of a recent commit that added blackbox erase. This is due to the limit of a uint_32 being reached in the MSP code that packs the "boxes" to send over uart...

borisbstyle, 2017-03-24:
I think the modes can be extended wirmth another  uint32_t parameter with some effort.

But also there are some unused modes like VARIO and perhaps some others too

ledvinap, 2017-03-24:
@borisbstyle : `uint64_t` is even easier ... at least in short-term

quiljw01, 2017-03-24:
For troubleshooting- it may be useful to add a graphic on OSD that shows when "Lava mode" is taking control.   Or maybe show on OSD "max gyro error" in degrees.  (this is the term that if it gets too high then Lava Mode gets enabled.  

Also the same bit could be passed to BB recording log.  

Maybe also slightly useful, but not nearly as much  - to have graphic on OSD showing when "Lava mode" is enabled.  

Is it useful to fire Lava Mode based on an error acceleration?  (derivative of error)  Or maybe total error over the previous time frame (Integral of error)  Probably a simple error based on single measurement could be reliable enough but these are ideas to test if the simple method does not work reliably. I might be getting ahead of myself.

quiljw01, 2017-03-25:
I found an example video clip showing where i think Lava Mode would save a ground impact.  The FC and motors/ physical system would have plenty of time to react to become perfectly level before contacting the ground.    https://youtu.be/p1T3dXnjido?t=15s

martinbudden, 2017-03-25:
Do you have the blackbox log for that crash? If so, can you attach it?

carloscampos775, 2017-03-25:
This is an excellent idea.

quiljw01, 2017-03-25:
I'm sorry @martinbudden  I do not have BB for this crash. However i can enable BB at minimum required freq permanently.   What is the minimum frequency required to have a useful BB view of the crash?   Also would be useful to clear BB log with stick command so i don't have to worry about storage filling up and missing the crash.

mikeller, 2017-03-26:
@ledvinap: In terms of not breaking MSP compatibility, adding a second `uint32_t` might be preferable. But I believe there is a fair number of modes that are unused, and have been so for a while, so it should be possible to repurpose them.

> Also would be useful to clear BB log with stick command so i don't have to worry about storage filling up and missing the crash.

@quiljw01: We do do that: #2267 :-)

ledvinap, 2017-03-27:
@mikeller : On MSP, only used bits are sent, so there is no need for change until 32 features may be used simultaneously.
Internally, enum is used as bit index, uint64_t will extend possible range quite safely (some places will need caser/LL suffix, but that should be hidden in feature-related macros and functions).

quiljw01, 2017-04-10:
Will Lava mode work while "3D mode" is enabled in "Config tab" and disabled with mode switch?  I don't need lava mode while 3D is active, only when it de-activated.

kmitchel, 2017-07-13:
@martinbudden Any tips for tuning threshold? With superexpo around ~83 it will not let me snap roll.

PeteG-FPV, 2017-07-25:
Have been testing this out, it's working so good, about 30 hits with gates now and it intervened on 28. On the other 2 occasions I went into the infinite yaw spin. Wasn't able to get BB log of it, but will endeavour to do so.

Scarl5, 2017-11-16:
im new at this and i dont want to use this feature, how can i disable it?

Scarl5, 2017-11-16:
@richard-scott but it does go crazy when landing sometimes

wind0r, 2017-11-16:
do you have airmode enabled?

Scarl5, 2017-11-16:
Yeah

wind0r, 2017-11-16:
then please read about it in the wiki or any other please.
your "issue" is normal behaivor.
please do not blame any features without knowing them.
as you see that doesnt help. people will just tell you how to disable it but you still have the "problem"..
just descripe the problem and do not assume it is a feature fault just because it is new

Scarl5, 2017-11-16:
I told at the begining im new, also i read things about the crash recovery and somewhere i dont remember but i read it was because of this feature, im not blaming on this feature. What it happens is that when i land and arm again motors start going at full thottle and yaw spin.

wind0r, 2017-11-16:
"I told at the begining im new" thats why i wrote it.
I perfere that newbes descripe there problem to find help so they can get pointed to the artictels or they even find the articels on there own.

e.g If your car doesnt start you would not ask how to turn down the radio or how to change the tires. You would descripe a mechanic your problem and he may can help (no fuel and so on)

You land, then disarm and then arm again and the quad goes full throttle?
And on a normal first arm  quad starts normal?

Please create a Issue since this was a Feature Request.

Scarl5, 2017-11-16:
Not full thottle sorry, full yaw spin

Scarl5, 2017-11-16:
And yeah first arm is allways normal

closed 2018-04-21 by DieHertz