betaflight/betaflight #2738: [Feature Request] Modes Selection

jgonzi, 2017-03-24:
Hi every, 

Always first many thanks for the great job you do. 

I want to know if we can have the option in mode selection instead of when we Add a Range to be an OR to be an AND. 

With the AND option with the combination of 2 switches we can have 4 different options, now with we OR option I cannot do it. 
[image]

I hope that idea could help.

Linjieqiang, 2017-03-25:
I don't agree more this great idea.

RipperDrone, 2017-03-25:
Why don't you create a mix on your remote to achieve 2 switches - 4 settings mix? This unloads further code from the FC...

jgonzi, 2017-03-25:
Well it is only to have down the mode the selection of a AND or an OR paramater. When FC has to decide it is two check... people who have for example a transmiter of 6CH will get more options.

dthubereng, 2017-11-30:
I'm interested in this feature as well. It seems to me this could be accomplished by being able to specify the state of the mode (ON or OFF) when the range condition(s) are met. Basically adding a boolean NOT to the output. This would effectively allow you to configure any mode with an AND'ing of ranges.

Better than this, but tremendously more code-intensive, would be the ability to mix ANDs and ORs in a sort of boolean statement using the ranges.

dthubereng, 2017-12-01:
Hmmm, after looking at the code a bit, this doesn't seem to be very straightforward. The issue, I think, is the fact that mode activations are stored by individual conditions (e.g. is the condition active? set the mode). This only allows for an ORing of conditions. Will continue to dig and think on this; surely there is an elegant solution.

This will of course require coordination with the configurator eventually, but I'm assuming interfacing through the CLI is also adequate for testing without it.

mikeller, 2017-12-01:
Yes, doing new features with CLI only support first is then normal way of doing things. This is good enough for testing, and allows eventual required fixes to be applied before the changes to the MSP protocol are made (which often happens in a second pull request against the firmware).

Once at that stage, we are happy to help with the configurator integration.

fiveangle, 2017-12-09:
Much better wording that my piss-poor badly-worded request for the same at https://github.com/cleanflight/cleanflight-configurator/issues/101 👍

closed 2017-12-13 by mikeller

trintcole, 2018-12-12:
I would like to see the AND function.  With realpit devices this could better setups to have multiple quads on one model.

PR #4722 (2017-12-13): Added AND Logic to Modes; Fixes #2738.
Added new configuration for storing mode logic type (AND or OR).
Added new 'mlogic' command to CLI. (is there BF documentation for BF-specific CLI?)
Added new bit array function for inverting a bit (XOR).
Updated code for determining the activation of a mode.
Fixes #2738
commit f9feb03a (2017-12-13): Merge pull request #4722 from dthubereng/master

Added AND Logic to Modes; Fixes #2738.