betaflight/betaflight #2744: BF angle/horizon strength value range

funkjan, 2017-03-25:
Hello, 

my first participation here - so I apologize for any 'form faults' :)

Need for my girlfriend my TBS colibri (RotorX Atom V3) to have a real angle mode - why real, because right now with a 'strength value' of 50 there is a real delay in the auto level reaction in angle mode - way too slow compared to other experiences.

That why I raised in Profile 2 the strength value first to 80 and meanwhile to 150, but it happens nearly the opposite than expected...!!

The quad is even slower returning to level???

Even if I assume that BF is not primarily made for autolevel modes, Is there any known bug concerning the values? or what could be the reason??

Appreciate your help!

Jan

enricoros, 2017-03-25:
The PIDs also affect the angle mode leveling speed, especially high P and D components will influence the auto level reaction time.

Try tuning your quad with higher PDs and see if it helps you more than the 'strength' parameter.

funkjan, 2017-03-25:
That is unfortunately not true anymore - with good tuned acro pid's there should be no reason to touch them anymore  - with these new versions of BF only these 2 parameters angle/horizon strength and transition should be necessary to change the behavior!!

There are no separate PID's anymore for the different modes and why play around with perfect acro PID's just to achieve faster responding Auto level??!

See here for ex.: https://www.rcgroups.com/forums/showpost.php?p=36366821&postcount=15

So still not solved - regarding the values strength and transition for angle/horizon 

Thx anyway!!

enricoros, 2017-04-07:
@funkjan see the following in the code:

1. In Angle mode with strength = 50, the "pidSetPoint" is set to 5.0*error_angle, where error_angle represents the difference between where the angle IS and where your RC remote wants it to be. See it here: https://github.com/betaflight/betaflight/blob/master/src/main/flight/pid.c#L278 

2. then the pidSetPoint gets fed into the PID loop and gets actuated using the PIDs - so low PIDs = low angle response as well. See it here: https://github.com/betaflight/betaflight/blob/master/src/main/flight/pid.c#L319

The "50" constant (which stands for 5.0) is for converting the "error angle" into "angular speed" - so 12 degrees of Roll error would be converted into 60 degrees/second of roll speed. Which means that if the PID are good (especially P is adequately high), it should get there in 1/5th of a second.

If, as you say, it takes longer, the Ps are probably low, and the code never lies, so trust the code and see that the PIDs you're using for acro are going to be used for level as well, with Strength converting from error angle to angular speed. When you fly Acro, the "pidSetpoints" are huge - you're asking for rates of 700 degrees/second for example, to do ~2 rolls/second so high rates can easily mask up low Ps in your PIDs (compare 700 deg/sec to the 60 deg/second you ask for a 12 degrees roll).

To sum it up: with "50" and a good tune you should perform any Angle change in 0.2 seconds or less.