betaflight/betaflight #2746: FEATURE ENHANCEMENT  | vbat min value alarm delay.

haybeseret, 2017-03-26:
Hello,
I am using the vbat alarm in addition to other telemetry option to monitor my vbat value.
When i am doing vbat alert on the Taranis for example i add a 3-4 seconds delay so it will not alert me on every punch out i am doing.

I would think it can be very good option to add it to BF as well so we can add a delay to vbat min alert , where the default will be 0.

mikeller, 2017-03-26:
Have you tried setting `vbat_hysteresis`? Instead of just delaying an alert, this allows you to configure how much sag you want to allow during a punch out.

haybeseret, 2017-03-26:
I tried playing with it but it was not too intuitive. I will try to search the documentation again, maybe i will get better understanding of it.
Still all i am aiming for is a state where vbat value is below or equal "x" value for "y" seconds then i will get the vbat alert.

henfra, 2017-05-03:
@mikeller
Might be I'm a little stupid but I couldn't find any useful resource on vbat_hysteresis. [Boris' explanation in rcgroups](https://www.rcgroups.com/forums/showpost.php?p=34454927&postcount=24700) wasn't helpful either.
I don't understand what it does, I looked at the code but my coding skills are even worse than my skills of finding any documentation on this xD

From the code (battery.c#L210-237) I'd assume that -- just considering voltage drop (BATTERY_OK->BATTERY_WARNING->BATTERY_CRITICAL) -- there is no difference between setting a lower warning voltage (e.g. 3.7V) instead of having a higher warning voltage (e.g. 3.8V) in combination with hysteresis (e.g. 0.1V). Only when "moving back up again" (BATTERY_CRITICAL->BATTERY_WARNING->BATTERY_OK) the hysteresis would come into effect (again: just in comparison of having set a lower warning voltage from the beginning)? I.e. the beeper is turned on later but turns off faster with hysteresis than without? But that still would make it beep on voltage-sag, it just would turn off faster after recovery of the battery?! That doesn't seem to make too much sense to me, but as I said, I'm probably misinterpreting the code...

ATM I have old batteries and the beeping is driving me nuts...a time-based delay (vbat below x for y seconds) would be great IMO.

Also: this seems like a duplicate of [issue 1887](https://github.com/betaflight/betaflight/issues/1887)

mikeller, 2017-05-12:
@henfra: As I said in #1887, a configurable delay for battery warnings would be a nice addition. Somebody will have to implement it, and I haven't got around to doing it yet. The hysteresis is a different kind of beast, it _will_ beep on punchouts, but on the plus side, it _will not_ delay letting you know when your batteries are getting really low.

Duplicate of #1887.

closed 2017-05-12 by mikeller