betaflight/betaflight #2749: Barometer dead in 3.1.6? (SPRF3)

Arakon, 2017-03-26:
I got an SPracing F3 clone here that used to work fine in 3.0.1 with the onboard baro. I'm just using it to display altitude on OSD.
After updating to 3.1.6, the baro is still detected, but always showing 0, both in the configurator and in the OSD.

jflyper, 2017-03-27:
@Arakon It works here with SPRacing F3 here (authentic, tested with bare FC).

- What if you go back to 3.0.1?

Arakon, 2017-03-27:
Hrm.. has to be some sort of config issue. I went to 2.7.1, no issues. Went to 3.1.6 with default settings, it works. Restore my CLI dump, baro instantly dead.
Setting baro hardware manually makes no difference.

`
# dump
# Betaflight / SPRACINGF3 3.1.6 Feb 21 2017 / 14:44:08 (1fd502c)
name Tweaker
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00
mixer QUADX

mmix reset

servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
[... 394 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 110
set rc_rate_yaw = 92
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 77
set pitch_srate = 75
set yaw_srate = 75
set tpa_rate = 10
set tpa_breakpoint = 1750

# `

Arakon, 2017-03-27:
Figured out what triggers it.. as soon as I disable the ACC, the baro also stops working. ACC on (doesn't need to be active via mode, mind), baro sends data.

mikeller, 2017-03-27:
That's weird. The original SPRACINGF3 doesn't do that. But then, there are a lot of clones of this board out there that have differences in what MPU is used, or how it is connected. What does CLI `status` tell you for your board (with ACC and BARO on)?

jflyper, 2017-03-27:
@mikeller @Arakon Confirmed with the authentic SP Racing F3 that disabling ACC causes BARO reading to be zero (still detects MS5611).

Arakon, 2017-03-27:
With ACC:

> # status
System Uptime: 4 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 1, config size: 1628
CPU:24%, cycle time: 262, GYRO rate: 3816, RX rate: 49, System rate: 9

Without ACC:

> # status
System Uptime: 19 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, BARO=MS5611
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 1, config size: 1628
CPU:13%, cycle time: 256, GYRO rate: 3906, RX rate: 49, System rate: 9

jflyper, 2017-03-27:
@mikeller It's in the `altitudeHoldGetEstimatedAltitude`. I think `EstAlt` becomes always zero if acc is turned off.

mikeller, 2017-03-27:
```
   if (sonarAlt > 0 && sonarAlt < sonarCfAltCm) {
        // the sonar has the best range
        EstAlt = baro.BaroAlt;
    } else {
        EstAlt = accAlt;
    }
```

This looks fairly terrible.
The fact that `baro.BaroAlt` is used in most places (telemetry, ...) instead of `altitudeHoldGetEstimatedAltitude` explains why this bug hasn't been reported until now.

My suggestion is: Fixup `altitudeHoldGetEstimatedAltitude` so that it properly gets the sonar / baro / acc altitude estimate, rename it to `getEstimatedAltitude`, and replace access to `baro.BaroAlt` with it.

Happy to do it, unless you want to.

Arakon, 2017-03-28:
Dirty fix that seems to work for me for now:
In fc_msp.c:
Replace 
```
#if defined(BARO) || defined(SONAR)
        sbufWriteU32(dst, altitudeHoldGetEstimatedAltitude());
#else
        sbufWriteU32(dst, 0);
#endif
        sbufWriteU16(dst, vario);
        break;
```
With 
```
#if defined(BARO)
        sbufWriteU32(dst, baroCalculateAltitude());
#else
        sbufWriteU32(dst, 0);
#endif
#if defined(SONAR)
		sbufWriteU32(dst, altitudeHoldGetEstimatedAltitude());
#else
        sbufWriteU32(dst, 0);
#endif
    sbufWriteU16(dst, vario);
        break;
```

That's probably not the right way to go about it (I'm no coder), but it does cause the baro to keep updating on the sensors tab and likely OSD even with ACC off.

mikeller, 2017-03-28:
@Arakon That would work. Even simpler, in the case of BARO being defined, we could just use baro.baroAlt, like most other places do.

But I will look into cleaning this up tonight.

PR #2760 (2017-04-10): Reworked altitude estimation. Made it work when ACC is disabled.
Fixes #2749.
closed 2017-04-10 by mikeller