betaflight/betaflight #2750: Loosing control at low throttle/airmode

slamblam, 2017-03-26:
Hello,

This is the second time this has happened... I was able to fix it last time, but this time the frame was destroyed and it snapped in half. When i have airmode engaged and when flying at low throttle for acro the quad will freak out or something happens with the gyro and it comes tumbling down. It does not do it every time it is as if the gyro disengages or something I do not get it. I had 3.1.6 installed the first time It lost control so I revertted back to 3.1.5 and it happened again.....  The other quad, with identical setup except with oneshot125, instead of dshot600, i upgraded it to 3.1.6, it did the same thing, lost control and came tumbling down. I reverted back to 3.1.5 and it seems to be ok...can you someone please look at the diff/dump below and see if I have something wrong....

# version
# Betaflight / AIRBOTF4 3.1.5 Feb  7 2017 / 21:59:59 (4646f9d)

# name

# resources

# mixer

# servo

# servo mix

# feature
feature VBAT
feature MOTOR_STOP
feature TELEMETRY

# beeper

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 5 33 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1250
aux 1 27 1 900 1150
aux 2 1 1 1875 2100
aux 3 2 1 1625 1750
aux 4 12 4 900 1325

# adjrange
adjrange 0 0 3 900 1225 6 2
adjrange 1 0 3 1375 1625 7 2
adjrange 2 0 3 1800 2100 8 2

# rxrange

# rxfail

# master
set min_check = 1010
set rssi_channel = 12
set fpv_mix_degrees = 25
set digital_idle_percent =  8.000
set 3d_deadband_throttle = 0
set motor_pwm_protocol = DSHOT600
set serialrx_provider = SBUS
set align_board_yaw = -90
set deadband = 6
set yaw_deadband = 6
set pid_process_denom = 1

# profile
profile 2

[... 480 lines cut ...]
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 86
set rc_rate_yaw = 86
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 88
set pitch_srate = 88
set yaw_srate = 88
set tpa_rate = 10
set tpa_breakpoint = 1650

frostbytedata, 2017-03-31:
Yea I was seeing this as well on my build. Only difference is, I have it during very low throttle, and max throttle.

davidsmd, 2017-04-01:
Seeing a similar issue here as well, but I'm relatively new to this, so it may be user error. Manifests in airmode or normal rate mode; quad is hovering at low-mid throttle, then rolls/pitches hard to one side into the ground. Have seen a few that look like a single motor is the culprit, and a few that involve two motors on the same side. 

Might also be related to a different issue in blheli_s regarding desyncs and software reversed motors. Will update after I try to get that sorted.

3.1.6 on Flip32 F4 piko/airbot f4 nano, 2 in 1 20a bb2 on blheli_s 16.6, dshot600

PAR2883, 2017-04-07:
Hi  please please I need help. so I seem to have a problem with my Tiny Whoop, I seem to lose link to TX mid flight. Well I believe its something like this.
I will be happily flying around no problems then all of a sudden I will just go forward and have no control of any yaw, pitch, roll or throttle. then once it crashes and comes down the motors are stuck on for a period of time. It will also do this and get stuck on a yaw spin then motors will just keep spinning. 

Please could someone help, I don't believe its my Spektrum DX6 as I seem to be flying my Robocat270 and Neon250 ok from it.
I just don't know what to try.

any help would be much appreciated.