betaflight/betaflight #2751: Funmode 1

enricoros, 2017-03-26:
Hi, as a feature idea it will be nice to have the following:
 1. when a particular condition is active (box defined in configurator),
 2. the Throttle gets modulated based on the _cosine of the angle with the vertical vector_ (and always > min_throttle)

This will enable fast rolls when the throttle is automatically modulated between 0 and the Rx level depending on whether the propellers are facing up.

With this fun-mode, people could do minutes of rolls and flips without losing altitude. Currently people 'cut' the throttle as the roll starts to avoid barrel rolls (and consequently the quad falls), or modulate the throttle by hand but only on low speed acrobatics.

Of course this requires the accelerometer to be present and on, but it could open the possibilities to new fun ways of flying.

jflyper, 2017-03-30:
I think cosine modulation doesn't work as you described; a vehicle will still loose altitude while not level (especially between 90-270 deg where I presume motors are kept at min).

varkokonyi, 2017-04-04:
It will lose altitude, but it will be easier to do lots of flips/rolls

enricoros, 2017-04-07:
It will lose altitude, but if rolls are fast and props spin higher when the thrust vector is facing up, the loss can be minimal and the quad will seem to "float up" :)  I'm sure even experienced acro pilots will find way to make new super fun manouvres and tricks!

fftunes, 2017-04-07:
Good luck doing a simple turn in this mode, as well as pairing use of accelerometer with hard acro...

enricoros, 2017-04-10:
Oh yes, this is only for fun, with a temp switch 💃.