betaflight/betaflight #2755: Feature: temporary store mAh used while rebooting after saving in CMS

basdelfos, 2017-03-27:
This weekend I did some tuning of a new quad with BFF3 flightcontroller. That means that during one lipo, I land frequently, adjust PIDs via CMS and test the tuned parameters. But after every change via the CMS (in this case onboard BF OSD) the FC reboots after the settings are saved.

Every reboots resets the mAh used to 0. Having to remember the last used mAh and adding it up in my mind to prevent over discharging the lipo.

This could be easely fixed by temporary storing the used mAh before the CMS reboots the FC. After rebooting restore the mAh used and clear the stored value. This way the used mAh is not cleared while rebooting on the same lipo. This doen't effect when the lipo is disconnected, than the mAh used should start at 0 when a new (other) lipo is connected.

jflyper, 2017-03-30:
Do you have any idea on how rebooted software detect whether the lipo has been changed?

basdelfos, 2017-03-30:
The idea was that when the FC reboots from CMS menu, it stores the mAh uses temporary and after the reboot restores the value again (and resets the stored value to 0). On lipo change the stored value will be 0 and therefor just resets it to 0. Only problem is when the user disconnects the lipo while the FC is rebooting from CMS, but that is unlikely to happen.

@mikeller suggested a better solution, actually this is just a workaround. The CMS menu now always reboots the FC regardless of the change that has been made. PID, filter, rates (and more) changes don't need a FC reboot to be effective. Better solution would be to only reboot after changes that has been made that actually need to reboot to be effective.

I'm currently working on a PR to implement this.

jflyper, 2017-03-30:
I have a code that only stores configuration (without rebooting) in the vtx consolidation code...

basdelfos, 2018-05-07:
This issue is gone obsolete, because there now is a Save + Reboot and only Save + Exit menu option in the OSD menu.

closed 2018-05-07 by basdelfos