betaflight/betaflight #2756: Hott on Naze not working on 3.1.6?

maylibu, 2017-03-27:
I was not able to get Hott working on a Naze 32 rev 6 in 3.1.6, over soft serial 1, pins 5/6.

I did get it working on 3.0.1, 3.1.0, 3.1.3, and 3.1.5.

For testing I just flashed the build erasing all data, and only enabled telemetry, soft serial, and selected Hott on soft serial 1. Using sumd on the receiver but don't think that matters.

Other details: GR-16L rx, MZ-12 tx, default oneshot125 selected for escs.

maylibu, 2017-04-04:
Hott telemetry on soft serial 1 with naze32 rev6 is also not working in 3.1.7.

wildflyer, 2017-04-05:
I have naze 32 rev 5 and the same problem. telemetry not working anymore in new version!!

wildflyer, 2017-04-05:
ok, I found out the problem, a difference between v 3.1.5 and 3.1.7
in the "resource list" in cli,  

"SERIAL_TX 11" is not assigned to "A07" (in version 3.1.5 it is assigned, I think that is the bug!)

how can I assign "SERIAL_TX 11" to A07 ? (so that it is shown up in resource list ?)
greetings

joachim

mikeller, 2017-04-05:
I had a quick look, and I can confirm this. Assigning SOFTSERIAL1 to a different serial function (e.g. MSP) causes both SERIAL_TX 11 on A07 and SERIAL_RX 11 on A06 to be assigned.
@jflyper: Could it be that for some reason in 3.1.7 telemetry uses the serial Rx pin only even if the protocol isn't bidirectional?
@wildflyer: Can you check if telemetry happens to work if you connect the RX telemetry pin to the SOFTSERIAL1 Rx pin (6)?

wildflyer, 2017-04-05:
hi,
the board is glued and soldered on the frame, I cannot access other pins than those that are already soldered. 
pwm5 and pwm 6 (on naze that is/was softserial1 rx and tx) are soldered together with a diode and connected to the gr-16 receivers telemetry pin.
in version 3.1.5 telemetry works well with these pins, in version 3.1.7 it is not working anymore and in cli "resource list" A07 is "free" instead of being assigned to  "SERIAL_TX 11"

greetings and thx for help!

joachim

mikeller, 2017-04-05:
Sorry, I wasn't aware that HOTT is actually a bidirectional protocol. In this case it is all as designed, and just connecting SOFTSERIAL1 Rx to the RX telemetry pin is all you need. Unfortunately, the now unused SOFTSERIAL1 Tx pin is set to high, which, together with the diode, causes is to interfere with the serial communication. So unless you unsolder / cut the connection between Tx pin and the serial line, or build a custom firmware that does not pull this pin to high, this won't work.

wildflyer, 2017-04-05:
so I  dont need the diode anymore ? are you sure ? so I just have to remove the diode and solder only 1 wire from pin 5 (pwm 5 /softserial rx) to the receivers telemetry pin ?

mikeller, 2017-04-05:
Yes that is correct.

maylibu, 2017-04-06:
Just tried this (single wire from the receiver's telemetry pin to pin 5 (softserial rx) on the naze32rev6) with no luck on 3.1.7. 

Confirmed it still works on 3.1.5 with the extra diode on pin 6/5. 

Should note it does not work for me on 3.1.5 either if I have the accelerometer enabled in the configurator along with the auto-level modes setup on an aux channel, with cpu usage reporting at 30-40%.

mikeller, 2017-04-06:
Your last remark makes it sound like MCU load is an issue. Have you tried if it works if you decrease loop frequencies?

@jflyper has done the softserial rework, he is probably in a better position to help you debug.

wildflyer, 2017-04-06:
so it doesnt work without diode ? what to do to get 3.1.7 working with hott telemetry ?

jflyper, 2017-04-07:
@mikeller 
Hmmm... `hott.c` currently implements half-duplex by explicitly switching between `MODE_RX` and `MODE_TX` by calling `openSerial` and `closeSerial`. This is why `serial_rx 11` is assigned at the beginning. I don't have any hott device, but you can probably observe resource assignment alternate between `serial_rx 11`and `serial_tx 11` if it is working.

I'm not so sure about the effect of the `hott.c`'s way of implementation on the current softserial implementation; never tested before...

mikeller, 2017-04-07:
Yeah, good point, that implementation should probably be changed to work properly with the new softserial code.

wildflyer, 2017-04-07:
if you need someone to test, maylibu can test a version without diode, and I can test a version with the diode. 
without would be better :-)

maylibu, 2017-04-08:
Sure, let me know if there is a .hex I can test.

sansha, 2017-04-08:
hi, I have the exact same problem.
Naze, softserial 1, Hott and SumD receiver, CPU load is at ~10% so I think this shouln't be an issue.
I also have my quad with the diode and could test a version. :)