betaflight/betaflight #2762: betaflight3.1.1 stuck on reboot

kiwikiwibob, 2017-03-28:
So I've been trying for the last 3 days to resolve this issue myself. I got my first fpv drone earlier in the week and I'm trying to config it. I have an Eachine wizard x220 with the spracing f3 fc on deck. I can connect and flash no problem, but when I try to save and reboot on ANY tab at all, it gets stuck on rebooting. I've updated my drivers, uninstalled and reinstalled chrome and betaflight. I haven't been able to find any threads on anyone else having the issue, and I would just like some help to get my first drone up in the air. Thank you all in advance.

jflyper, 2017-03-29:
I'm afraid we can't help you because we don't have the hardware (the fc, in particular).

cjmack03, 2018-03-12:
I've been having the same exact problem. This is my first racing drone and was stoked to get it flying. Now I can't use it until I solve this problem

cjmack03, 2018-03-14:
I figured it out. I also have the Eachine Wizard x220. The reason this is happening is because your boot jumper is still in place. Once your quad has finished flashing you can remove the jumper and connect. It should save and reboot without a problem after that.