betaflight/betaflight #2764: BF 3.2.0 - rapidly crazy yaw spin

TheAngularity, 2017-03-28:
today i wanted to test again bf 3.2.0 ... but unfortunately i could not do this ... after arming and some throttle the quad comes to full throttle and rapidly crazy yaw spinning ... absolutely uncontrollable ... only disarming stops this wild rodeo ... someone any idea

here my diff:
```
# Betaflight / REVOLT 3.2.0 Mar 28 2017 / 14:49:40 (c39ab90)

# name
name HX200

# feature
feature VBAT
feature LED_STRIP
feature AIRMODE

# beeper
beeper -ON_USB

# map
map TAER1234

# serial
serial 5 64 115200 57600 0 115200

# led
led 0 11,10::AW:0
led 1 10,10::AW:0
led 2 9,10::AW:0
led 3 6,10::AW:0
led 4 5,10::AW:0
led 5 4,10::AW:0
led 6 9,5::CW:6
led 7 8,5::CW:6
led 8 7,5::CW:6
led 9 6,5::CW:6
led 10 6,2::CW:6
led 11 7,2::CW:6
led 12 8,2::CW:6
led 13 9,2::CW:6

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1600 2000
aux 1 12 1 1000 1200
aux 2 14 1 1000 1600
aux 3 25 0 1600 1800

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set moron_threshold = 100
set acc_hardware = NONE
set min_check = 1000
set serialrx_provider = SUMD
set blackbox_rate_denom = 8
set blackbox_device = SPIFLASH
set digital_idle_percent =  7.000
set motor_pwm_protocol = DSHOT600
set battery_meter = ADC
set vbat_scale = 109
set pid_process_denom = 1
set ledstrip_visual_beeper = ON

# profile
set d_notch_hz = 0
set pid_at_min_throttle = OFF
set p_pitch = 64
set i_pitch = 46
set d_pitch = 24
set p_roll = 56
set i_roll = 42
set d_roll = 26
set p_yaw = 80
set i_yaw = 55

# rateprofile
set rc_rate = 110
set rc_rate_yaw = 105
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 75
```

borisbstyle, 2017-03-28:
Thanks for reporting. 3.2 is really not much flight tested at all.
Any chance to get blackbox log of it?

TheAngularity, 2017-03-28:
I had forgotten ... here it is

[BLACKBOX_LOG_HX200_20170328_193756.TXT](https://github.com/betaflight/betaflight/files/876847/BLACKBOX_LOG_HX200_20170328_193756.TXT)

TheAngularity, 2017-03-29:
tested now ... all working normal again ... #2768 fixed this issue

closed 2017-03-29 by TheAngularity

mikeller, 2017-03-30:
@TheAngularity: The issue in #2768 was that the yaw input was reverted. I wonder how this made your craft 'rapidly crazy yaw spinning ... absolutely uncontrollable'. ;-)