betaflight/betaflight #2765: Switchable osd layouts

alessandroame, 2017-03-28:
I would select from 3 osd layout presets via an rc switch for:
-pid tuning
-freestyle
-race

mikeller, 2017-03-29:
If you are using MWOSD, this can be done to some extent (only 2 settings) by using the OSD_SWITCH switch.

DanNixon, 2017-03-29:
@jflyper Do you still have plans for this?
If not I could take a look.

jflyper, 2017-03-29:
@DanNixon As we have completed the PG migration, we can start working on this. Do you have a plan? (We need a substantial configurator change also.)

DanNixon, 2017-03-29:
The configurator change was the only thing I am uncertain on.

My solution would require essentailly multiple OSD switch boxes which I'd rather be able to represent on a single range rather than having a seperate range selector for each box (the box in thios case is what selects a particular screen).

The config end of it should be just a case of expanding the `item_pos` array to two dimensions (i.e. `[screen index][item]`).