betaflight/betaflight #2778: No smartport telemetry on BRF3 with 3.1.6

lijingwei0710, 2017-03-30:
connected uart 2 tx to xsr s.port
telemetry outport is SmartPort and AUTO
telemetry is enabled in configrtation tab
set  tlm_switch = on
set tlm_inversion = on
no new sensors found on telemetry page of taranis

jflyper, 2017-03-30:
Don't fiddle with CLI variables; just turn on SmartPort and TELEMETRY.
(Your configuration may work when armed.)

mikeller, 2017-03-30:
`set tlm_switch = on` this means that you have to assign and use a switch to explicitly disable MSP / enable telemetry on the port used for telemetry *and MSP*. If you're not sharing the telemetry port, enabling this makes no sense, and it's probably causing your config to not work.

As @jflyper says, first try the config that you can get from the GUI, only if that doesn't work fiddle with the CLI. We're trying to make the most common config the default, so that's what should work for most people.

johnnyruz, 2017-03-30:
Yeah, I can report no issues with SmartPort Telemetry on the SPRacingF3 on 3.1.6. I'm using an X4R-SB connected to UART3. Only had to enable SmartPort for the UART on the Ports tab and Telemetry on the Configuration tab.

closed 2017-04-05 by lijingwei0710