betaflight/betaflight #2779: [Feature Request] Prevent drone from skyrocketing when "side" hitting an object.

nitbeat, 2017-03-30:
This is just an idea I'm throwing and it will probably need some smart algorithm behind it.

Basically detect one motor stopping (when one of the sides of the drone hits something)
and shut down the diagonal motor, in order to prevent the racer from shooting up in the air,
and just make it continue strait forward.

I think what happens is the FC try to compensate for the yaw movement from the hit, and spools up the diagonal motor !

Thanks...

DieHertz, 2017-03-30:
It may be rather easy to detect such situation when more motors with eRPM telemetry emerge.

BayaroX, 2017-06-09:
This "issue" looks like this one >> https://github.com/betaflight/betaflight/issues/2731

closed 2018-04-21 by DieHertz