betaflight/betaflight #2780: CC3D Missing Receiver Channels After Updating From 3.0.1 to 3.1.6

dustinb, 2017-03-31:
This is a CC3D Atom Evo.  Backed up the configuration, flashed 3.1.6, restore configuration.

Yaw shows up as aux 1 and my flight mode switch shows as aux 2.   Channels 1-4 show no activity on the receivers page.  Will upload my backup.json when I get chance to zip it.

Downgrading back to 3.0.1 everything works as before.  Would like to upgrade as the CPU usage seems lot better.

McGiverGim, 2017-03-31:
I had the same problem with some version (I don't remember if it was 3.1.6), but with the latest code of the repository it seems solved. I don't know if you're able to build the firmware with the actual code to see if it is solved for you too.

McGiverGim, 2017-03-31:
I've found the Issue where we talk about:

https://github.com/betaflight/betaflight/issues/2637

Take a look at the description and the video, to see if it's the same issue ;)

dustinb, 2017-03-31:
I should mention I'm using PWM single channel per wire.  I know right.  I took a quick look at that issue but didn't understand how it related. I will review it in more detail.

[betaflight_backup_20170326_150254.zip](https://github.com/betaflight/betaflight/files/885897/betaflight_backup_20170326_150254.zip)

dustinb, 2017-05-07:
Tried release 3.1.7 and have the same issue.  Not seeing any activity for 1-4 channels, rudder shows on AUX1, AUX2 showing what was previously AUX1. Back to 3.0.6 I go

dustinb, 2017-06-29:
Closing, user error, ID10T, PEBKAC, etc..   After replacing the cc3d with a new spracing f3 controller and having the same issue I finally found channels 1-4 were in failover mode.   Possibly some change in default setting or 3.1.7 detecting lower pwm values? not sure but resolved.

https://www.rcgroups.com/forums/showthread.php?2601362-PPM-PWM-not-working-on-SP-Racing-F3-Not-registering-on-CleanFLight

closed 2017-06-29 by dustinb