betaflight/betaflight #2784: YAW I term limitation in 3D but also general

DietmarM, 2017-03-31:
In 3D the Yaw limit does not work:

// in 3D mode, mixer gain has to be halved
 385 if (feature(FEATURE_3D)) {
 386 if (motorCount > 1) {
 387 for (int i = 0; i < motorCount; i++) {
 388 currentMixer[i].pitch *= 0.5f;
 389 currentMixer[i].roll *= 0.5f;
 390 currentMixer[i].yaw *= 0.5f;

For me this seems to be obsolet because because scaling is done different in the mean time - this leads to motorMixRange never reaches 1 which causes no YAW I term limitation.

I wonder in PID.C also about this:

if (motorMixRange < 1.0f) {
 // Only increase ITerm if motor output is not saturated
 ITerm += Ki[axis] * errorRate * dT * dynKi * itermAccelerator;
 previousGyroIf[axis] = ITerm;
 }

should it not still be added if the new term reduces the current Ki[axis] value? Adding a negative numer to a positive reduces as well as adding a positve number to a negative the net value.

This might cause some never ending spinning copter (not highly propable because P can lower motorMixRange and help by that to bring the ITerm down-but depaning on PID setting and circumstands).

borisbstyle, 2017-03-31:
Yes you are right. That part is obselete and should go away. It all comes from the legacy code where there was a direct relation between pid scaling and motor outputs.
That is now not the case anymore in 3.1 and must be removed.

You are also right that windup should probably be allowed to decrease only during those scenarios.

martinbudden, 2017-03-31:
The change in `pid.c` allowing the windup to decease is in PR https://github.com/betaflight/betaflight/pull/2786

DietmarM, 2017-03-31:
(ITermSaturated == false || (-ITermNew > -ITerm && ITermNew < ITerm))

is that correct for negative values?
I guess you need to pathes Iterm <0 or Iterm >=0

borisbstyle, 2017-03-31:
@DietmarM Actually 3D mode does require halfing of the mixer othewise the PID's would be stronger in 3D.
Same pidsum in 3D means more powerfull correction otherwise. Halfing it kind of makes it equal.

DietmarM, 2017-04-01:
Yes you are right comparing the old code PID 3D only PID values  you have to be halfen. But comparing normal mode to 3D they should be more equal with the new code.

DietmarM, 2017-05-20:
On the sticking ITerm I like the solution with the Abs values - also tested it positive:

         // -----calculate I component 
425         const float ITerm = axisPID_I[axis]; 
426         const float ITermNew = ITerm + Ki[axis] * errorRate * dT * dynKi * itermAccelerator; 
427         const bool outputSaturated = mixerIsOutputSaturated(axis, errorRate); 
428         if (outputSaturated == false || ABS(ITermNew) < ABS(ITerm)) { 
429             // Only increase ITerm if output is not saturated 
430             axisPID_I[axis] = ITermNew; 
431         } 

Even I'dont understand still the multiplication:

closed 2017-05-20 by DietmarM