betaflight/betaflight #2799: BF 3.1.6 with SP F3 download blackbox log freezes

attilafustos, 2017-04-02:
Hi guys,

I have this issue with latest BF 3.1.6 and Seriously Pro F3 Acro.
I can log data but I cannot download the log to my PC. The progressbar freezes at a random location.
I tryed a new USB cable, I updated all drivers.
I have latest BF configurator.
If I downgrade to BF 2.9.0 I can download the log file but I cannot see the graphs in the viewer app.
I need this fixed so that I can start PID tuning.

Thanks,
Attila

mikeller, 2017-04-03:
What platform / OS are you running the configurator on? There is an issue with MacOS: https://github.com/betaflight/betaflight-configurator/issues/435

attilafustos, 2017-04-03:
Windows 7 x86

attilafustos, 2017-04-03:
I seen another guy complaining about this same issue on other site.. bountyhunter or what it was named

mikeller, 2017-04-03:
Thing is, it works fine for me with an SPRACINGF3, so I can't reproduce or test, and it's unlikely to be caused by a bug in the firmware / configurator. Is there any device connected to UART1 on your board?

attilafustos, 2017-04-03:
I have MWOSD and IBUS receiver

attilafustos, 2017-04-03:
MWOSD is on UART1

mikeller, 2017-04-03:
Can you try disconnecting the MWOSD and then downloading? USB is connected to UART1 too on SPRACINGF3.

attilafustos, 2017-04-03:
Yes, I will try that when I get home today. Great tip!

attilafustos, 2017-04-04:
works. thank you

closed 2017-04-05 by mikeller

attilafustos, 2017-04-08:
I double checked my ports today. I am actually using UART 2 for MWOSD. But last time I tryed downloading the log, If I disconnected the OSD worked...