betaflight/betaflight #2803: From BB425 to BF3.1.6 should I change buzzer

Eligiusz, 2017-04-02:
Hi,

I was using Raceflight on Revo from RTFQ and buzzer was working fine on motor 6 like below: 
[https://img.rcgroups.com/http://i.imgur.com/zhJXLBv.jpg?h=X0bszi16Tfqxn2TjjanG6w](url)

Now I want to move to BF but I have a problem with buzzer, I was trying #1736 but when I do:

```
# resources
resource BEEPER A00
resource MOTOR 6 NONE
```

buzzer is not working...

should I change this buzzer to the diferent one and change the connection? 
What I do wrong?

AndersHoglund, 2017-04-02:
That still works just fine for me.  Did you remember to `save` ?

Eligiusz, 2017-04-02:
Yes, do you have any other idea?

AndersHoglund, 2017-04-02:
No, no other ideas. Simple thing, should just work. Or, you might want to play with `beeper_inversion` ON or OFF. But you should hear something, regardless if inverted or not. Depends on what you mean by "buzzer is not working". Complete silence, or ??

Eligiusz, 2017-04-02:
yes, compleate silence

Eligiusz, 2017-04-02:
Maybe this will help:
`2017-04-02 @ 20:38:53 -- Running - OS: Windows, Chrome: 56.0.2924.87, Configurator: 3.1.1
2017-04-02 @ 20:39:18 -- Loaded release information from GitHub.
2017-04-02 @ 22:48:43 -- Serial port successfully opened with ID: 43
2017-04-02 @ 22:48:44 -- MultiWii API version received - 1.31.0
2017-04-02 @ 22:48:44 -- Flight controller info, identifier: BTFL, version: 3.1.6
2017-04-02 @ 22:48:44 -- Running firmware released on: Feb 21 2017 14:37:31
2017-04-02 @ 22:48:44 -- Board: REVO, version: 0
2017-04-02 @ 22:48:44 -- Unique device ID received - 0x22001f3335510b34363336
2017-04-02 @ 22:48:53 -- CLI mode detected
Setup
Ports
Configuration
PID Tuning
Receiver
Modes
Motors
CLI
Note: Leaving CLI tab or pressing Disconnect will automatically send "exit" to the board. With the latest firmware this will make the controller restart and unsaved changes will be lost.

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# Betaflight / REVO 3.1.6 Feb 21 2017 / 14:37:31 (1fd502c)

# name
name -

# resources
resource BEEPER 1 A00
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
[... 449 lines cut ...]
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 112
set rc_rate_yaw = 112
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 81
set pitch_srate = 81
set yaw_srate = 75
set tpa_rate = 10
set tpa_breakpoint = 1650

# `

Eligiusz, 2017-04-03:
AndersHoglund please be so kind and send me your dump log from BF.

AndersHoglund, 2017-04-03:
Sure. Both a diff and a dump. [Mako_Revo_BF317_config_20170402.zip](https://github.com/betaflight/betaflight/files/891311/Mako_Revo_BF317_config_20170402.zip)

AndersHoglund, 2017-04-03:
For the record, this is the transistor stage I use. 
[image]

Eligiusz, 2017-04-03:
The last thing, could you please send me also BF 3.1.7?
I want to be sure that everything is like in your quad...

AndersHoglund, 2017-04-03:
3.1.6 or 3.1.7 should not matter much. But get it here if you want it: http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Maint/

Eligiusz, 2017-04-04:
Once more thanks :)
I will try to solve this maybe in the weekend...

AndersHoglund, 2017-04-04:
One more thought, you might want to try a pull-up resistor with your buffer circuit. Not sure what state the M6 output is in when re-mapped to beeper. Might be OD. My PnP circuit is not sensitive to that.

closed 2018-04-21 by DieHertz