betaflight/betaflight #2805: CC3D flexiport and mainport

Ndiw, 2017-04-03:
dear devs,
I don't know wether this is the configurator or firmware problem..
I tried to use both flexi and mainport as SBUS receiver and Bluetooth module for connecting to betaflight and using RD9 Radiolink AT9
I can use only one of the, for example use the sbus rx OR the bluetooth module, not both at the same time, so to be clear I only can use bluetooth but unplugged sbs connection and so the other way around
I get this working on librepilot 16.09 perfectly at 115200 baud with no problem whatsoever

thank you

mikeller, 2017-04-03:
Can you do a CLI `diff` please?

Ndiw, 2017-04-03:
i'm relatively new to betaflight, can u give me instructions?

Ndiw, 2017-04-03:
`CLI

# diff
# Betaflight / CC3D_OPBL 3.1.6 Feb 21 2017 / 14:16:31 (1fd502c)

feature -RX_PPM
feature RX_SERIAL
feature MOTOR_STOP
serial 0 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
set motor_pwm_protocol = OFF
set serialrx_provider = SBUS
profile 0

rateprofile 0
rateprofile 0

# `

mikeller, 2017-04-03:
```
serial 0 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
```

You have configured the MainPort (`serial 0`) for MSP (`1`), and the FlexiPort (`serial 2`) for Serial RX (`64`). SBus requires an inverted serial signal, and only MainPort on CC3D has an inverter (according to https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20CC3D.md). Try swapping them around, and using MainPort for SBus RX, and FlexiPort for MSP over BlueTooth.

Ndiw, 2017-04-03:
negative, still not working.. i tried swap physically and serial 0 2 in configurator, only receivers responding .. configurator on other laptop (I use 2 computers) with the bluetooth no data received, it says "no configuration received within 10 sec..." then port closed, the bluetooth module confirmed connected though, it blinked slowly indicating that bluetooth connection paired and connected

# diff
# Betaflight / CC3D_OPBL 3.1.6 Feb 21 2017 / 14:16:31 (1fd502c)

feature -RX_PPM
feature RX_SERIAL
feature MOTOR_STOP
serial 0 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
set motor_pwm_protocol = OFF
set serialrx_provider = SBUS
profile 0

rateprofile 0
rateprofile 0

#

mikeller, 2017-04-03:
I wasn't aware that bluetooth connections are supported by the configurator. How do you set up the bluetooth serial port in your OS, and what to you use as 'port' setting in configurator?

Ndiw, 2017-04-03:
please watch this video.. it's cleanfligt though, I assume betaflight was inherited from cleanflight, I might be wrong CMIIW
https://www.youtube.com/watch?v=aM9bq6J02jE

AndersHoglund, 2017-04-03:
BF and BFC should not care if they are connected via a wire or BT. But still, have you tried a wired direct connection, not using BT? 
I will see if I can duplicate this, hold on.

Ndiw, 2017-04-03:
wired meaning configuring the board with BF? I tried it on a pc desktop directly to cc3d via usb, i checked the receiver rensponse by looking at channels bar in the receiver tab on BF interface.. it responded, meaning that sbus is working, of course I "disconnect" the BF configurator to try to connect from other computer using a bluetooth, take your time sir, I will keep my eyes on this thread. thank you

AndersHoglund, 2017-04-03:
No, not the CC3D USB connector. I mean connecting a USB serial adapter to the same serial port as you have the BT device. Everything the same, only diff being BT wireless or USB-adapter wired. To the same FlexiPort.

AndersHoglund, 2017-04-03:
Some initial testing with 3.1.6 shows some issue with MSP on UART3, Flexiport. Does not connect. UART1 is fine. Same problem on current HEAD of patch_3.1.7. This has nothing to do with BT nor Sbus, only simply no MSP on Flexiport, UART3.
Could this the problem t you see too? If you can do a mod to your SbusRx, to make it send non-inverted serial data, you could run MSP on UART1 and non-inverted Sbus SerialRx on UART3.

Tried the same tests on REVO with BF3.1,7 and 3.2, MSP works on both UART1 and UART3 in both cases. So, I tried CC3D on BF3.2 too, and now MSP works on both ports. Hmm, strange.. will do some more investigations....

mikeller, 2017-04-03:
@AndersHoglund: I don't have the hardware, so I can't test, but my suspicion that it is caused by I2C sharing the pins with UART3. CC3D defines `USE_I2c` in 3.1.6, but doesn't in 3.2. Probably needs the same code as https://github.com/betaflight/betaflight/blob/master/src/main/target/NAZE/initialisation.c#L38-L40 .

Ndiw, 2017-04-03:
*fingercrossed*

Ndiw, 2017-04-03:
ok, this means that the patch at target.h by mikeller will be available in  3.1.7? cmiiw

AndersHoglund, 2017-04-03:
Jepp. Already there, in the Jenkins-CI alpha builds. Try this one: http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Maint/105/

Ndiw, 2017-04-03:
apparently... it works like a charm guys... 
my god, you're awsome guys.. 
I don't know how to thank you, beer? lol

Ndiw, 2017-04-03:
there seem tobe a small problem, but this is alpha, it's okay can live with it..
the link to bf seem tobe a little bit jerky, the "chain" icon on top below the battery icon, come off and on, but like i said, i can live with it. thanks again guys

VvIRONWOLFvV, 2017-04-04:
How do i go about deactivating 12c on uart3? I'm assuming this will fix my osd on flexiport not working

mikeller, 2017-04-04:
@VvIRONWOLFvV: Download a build with the fix from here: http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Maint/105/

Ndiw, 2017-04-04:
Try the alpha firmware 3.1.7 in the link attached above anders provide

VvIRONWOLFvV, 2017-04-04:
Thanx a million guys

closed 2017-04-05 by AndersHoglund