betaflight/betaflight #2812: rotorgeeks ssd f3 blackbox sd card "fatal error"

yosemitez, 2017-04-03:
bf  rg_ssd_f3   3.1.6 2/27/17 stable

new board, fresh install and kept getting "fatal error" when inserting sd card, yes bb is enabled, yes sd card is formatted, etc.  i noticed this in the dump;
feature -BLACKBOX   even though it is enabled
changed to    feature BLACKBOX  and now it records fine

funny thing is that also happened to my other rr ssd f3 board in a quad i have been flying for months after upgrading the fw, the bb had been working for months
this one read "feature BLACKBOX" but it was not working so i:
feature -BLACKBOX
feature BLACKBOX
and it started working

mikeller, 2017-04-03:
So is there anything that is not working now?

It looks like the BLACKBOX feature got disabled in your config for whatever reason. Re-enabling fixed this. Also, sometimes SD cards fail to properly initialise when they are first inserted into the flight controller after reformatting. Rebooting usually fixes this.

yosemitez, 2017-04-03:
all is working now, rebooting did not fix the issue

how would the bb get disabled on a new fresh install? or on one that had been working?

mikeller, 2017-04-03:
BLACKBOX seems not to be enabled by default on this target, so a config reset during a reflash might have caused this.

yosemitez, 2017-04-04:
thanks, just reporting in case other folks have the same problem,thanks so much for providing a way to have so much fun!

closed 2017-04-13 by yosemitez