betaflight/betaflight #2815: New RC Curve Implementation [Need GIT help]

nateimig18, 2017-04-04:
So I have code for a new RC curve implementation that uses the combined output from both a fast powf and expf routine that allows "ultimate control" over the remapping curve's max-rate, center-stick slope, center-stick linearity width, & a mid-stick non-linearity parameter.  Below is an example of a constant center stick slope & normalized max-rate plot of 30 curves with various widths, as there non-linearity parameter is swept 0.37 to 10.

[image]

This may seem impractical with the computational complexity however, I've already created the appropriate approximations for all the required routines with multiple order minimax polynomials.

Really the only issue left is what are the steps I should take to contribute this code of mine into this project using GIT (GIT Extensions)? Any help would be much appreciated!

mikeller, 2017-04-04:
Hi @nateimig18. Thanks for your effort to improve Betaflight.

For the question of how to use git to contribute to a project, there is a pretty good primer: https://help.github.com/articles/fork-a-repo/
(Essentially, create a fork, create a branch of your fork, do your code changes on this branch, when ready to be merged back, create a pull request from your fork's branch into `betaflight/master`.)

If you are not yet set up to build firmware binaries, then there is a number of ways of how that can be done, depending on what OS you are using. One way that is pretty much OS independent is to do your builds in a docker container: https://github.com/betaflight/docker-betaflight-build

DieHertz, 2017-04-04:
I'm sure someone could help you with tackling git if you join Slack:
betaflight.nz, betaflight.tk

nateimig18, 2017-04-04:
@mikeller & @DieHertz  Thanks for the quick responses! Unfortunately Im a little lost just trying to fork Betaflight, keeps on forking the cleanflight project?

I posted on the slack forum under #general

closed 2018-04-21 by DieHertz