betaflight/betaflight #2817: Quad flips at mid throttle

Miniwen, 2017-04-04:
Hello,

I'm having a very strange issue, with SP3 FC, on my 250 quad.
Quad flips when throttle input reaches ~50% (40-48% depending on battery charge).
If throttle increased instantly at take off it does instaflip, if throttle increased slowly it hovers (quite stable), but when throttle reaches 50% it falls down (always at the same manner - the same arm goes down). I cannot solve this for more than a month. replaced half of quads hardware... 
I add a black box log if it is any help.
[5.zip](https://github.com/betaflight/betaflight/files/893030/5.zip)

jflyper, 2017-04-04:
(Without looking at the log) Bad desync.

You can move the motor/esc on the arm that's going down to other arms to isolate the problem.

andre92, 2017-04-05:
You mean replacing hardware for new or swapping around between arms?
Looking at the log, it appears that motor 1 is dropping but throttle goes up for motor 3, is that correct?
Have you checked the motor order and direction with the motors tab? No propellers upside down? My best guess from the blackbox is props upside down, motors spinning in the wrong direction or two motor signal wires swapped. Spin motors up one by one, make sure everything is correct.
Also, I see you have it in Angle mode. Try lowering p_level to half the default, just to rule out a PID oscillation, or fly it in acro mode.
I would like to see your CLI dump and pictures or a video. Are you using a custom mixer by any chance?

jflyper, 2017-04-05:
@andre92 But @Miniwen says it hovers stably under some condition.

Miniwen, 2017-04-05:
By replacing I mean throwing old ESC away and buying a new ones.
Props are not up side down for shure, nor the morots spin the other way because it howers at low throttle.
I use no costum mixing. Everyting is disabled, eaven failsafe.
I was able to record hovering and droping black box log:
[6.zip](https://github.com/betaflight/betaflight/files/896299/6.zip)

I think as @jflyper  said this might have to do sth with desinc. Tried to google how to fix desinc problem, found nothing valuable. But managed to diagnose that when is instantly put from 0 to max (the same motor whose arm is faling down) in betaflihgt motors tab ir does not spin, but glitches. Switched esc's with the other leg,then that other motor glitches, so I asume this will be a bad ESC again?

Is there any way to fix this via BLHelli suite or do I need to buy a new set of ESC's and try again?

P.S. Tried to play with demag and motor timing - nothing helps.

mikeller, 2017-04-05:
Just to make sure: You have calibrated your ESCs?

robschii, 2017-04-09:
i have nearly the same issue   but at full throttle. quad flips because motor 1 has no power put  motor signal is 100%     problem is at bf 3.1.6 and 3.1.7     at 3.1.5 everything is ok

two diffrent setups with the same prpblem at 3.1.7

borisbstyle, 2017-04-11:
@robschii looks like ESC desync. Increase min_throttle or idle percent and see if that helps

robschii, 2017-04-11:
hi boris with 3.1.5 the problem is completly away.    at 3.1.6 the motor one sounds horrible after 2 minutes   and after some punchouts the motor one has no power any more.    with 3.1.7  it has no power from beginning.      i think desyn  will end in a fast flip or something else.     but it flip slowly at the beginning if you pitch backward slowly and than fast
i hope you can understand,    deutsch wäre mir lieber

robschii, 2017-04-13:
should desync start  from the beginning and not after some minutes of flight

trojanGoat, 2017-04-16:
Seems like I'm having a similar problem.  I already tried flashing 3.1.7 and 3.1.6.  I tried dshot600,300 and multishot.  Also tried increasing min throttle.  It almost seems random.  I tried using the motor tab to see if there's anything funny happening at mid throttle, it all seems fine.  Sorry no blackbox info this time around just these videos.  It always rolls to the left.  It also only happens when i'm flying forward.  It won't happen if I do punch outs.

https://youtu.be/4sF_z2w9ihY

For now I'll try removing software motor reverse.

Noctaro, 2017-05-03:
@trojanGoat For me, reducing the motor startup power in blheli solved the desync issue. Maybe it help for your build aswell.

trojanGoat, 2017-05-03:
@noctaro it turns out that one of those rectangular caps came loose on my ESC,  I guess that was causing it to desync.  My quad is all good now

KrackJack1, 2017-05-09:
@robschii  did you solve the problem? I have the same problem. I have changed my ESC and my motors but still can't figure it out... after flashing beta flight 3.1.7 on my omnibus f4....

Miniwen, 2017-05-21:
In my case - bad ESC. It sort of works, but not correctly. I changed it and some of the problems dissapered (others I can not test, because acidently fried reciever)

closed 2018-04-21 by DieHertz

whasianben, 2019-04-26:
I have the same issue. When I reach about enough throttle to take off, it flips out. I am using completely default settings