betaflight/betaflight #2820: Keep unchecked "disarm regardless of throttle value" alive!

quiljw01, 2017-04-05:
have a new request guys - as of right now - when 3d mode is checked in config tab - this setting is ignored : "disarm motors regardless of throttle value".  (config page)

I understand that IN 3D MODE it might be potentially dangerous to allow motors to continue spinning until throttle value goes to any specified stick value - so i am ok with this being the case while 3d mode is on. 

HOWEVER  - i would like this to change when "DISABLE 3d switch" is used in MODES tab.

When 3d disabled switch is active, (meaning that 3D mode is OFF )  i would like to disarm and continue flying until my throttle stick goes to minimum value.  This is like it does normally without 3d mode enabled.  

It is very helpful for landing to uncheck "disarm motors regardless of throttle value"... this way, once you're ready to land, on your approach to landing spot you can flip your disarm switch a few seconds early, then approach the landing spot, and as soon as you drop the throttle completely the quad falls and disarms in one motion.  I would like to continue being able to do this.  

Is this possible?  Can i have this please?

tavdog, 2018-03-09:
This this feature get removed from BF3.3 ?  I can't find it anywhere and I need it !

AndersHoglund, 2018-03-09:
Dangerous option removed. Read the 3.3 Release notes, last bullet in section "Important information when upgrading". https://github.com/betaflight/betaflight/releases

tavdog, 2018-03-09:
I guess I'm moving to Butterflight then.

DieHertz, 2018-03-09:
You are free to use whatever software you like.

voneiden, 2018-03-14:
I don't know if there's any significant demand on this, but just a side note, this could be implemented as an independent "auto-disarm" mode rather - if toggled on, throttle is low and armed then trigger disarm. That way it would never interfere with the actual disarm switch in case of emergency.

AndersHoglund, 2018-03-14:
There is yet another alternative available already today that is very similar to the way you guys land, and no new features or switches needed.  Richard Dell on FB posted this: 

> For all those complaining about losing this feature, you can try this: 
> In the cli find `pid_at_min_throttle`  and set it to OFF.
> Use a switch for air mode which will override this setting while air mode is active, when ready to land switch air mode off. As you touchdown with the throttle at idle the pid's will be ineffective so the quad won't bounce and so on. Problem solved. Then learn how to land......

mikeller, 2018-03-14:
This feature has been removed.

closed 2018-03-14 by mikeller

voneiden, 2018-03-16:
Alrighty. I've never used this feature myself, but it's good to have that tip by Anders here in case people land on this issue through google looking for answers.

SangaeLama, 2018-06-21:
Hey is there anyway to turn off this feature? "Disarm motors regardless of throttle value." I accidentally moved disarm switch in the air and disarmed my quad. I want to be sure it does not accidentally disarm on midair.
Thanks.

tavdog, 2018-06-21:
@SangaeLama The feature has been removed so you can't change it's behavior.  You can get around it by using a logical switch in your transmitter to only send the disarm signal if the throttle is low. If you have a taranis / opentx it's doable.