betaflight/betaflight #2826: Brushed motors spin to 100% when selecting the motors page on the configurator

theseankelly, 2017-04-05:
This didn't happen about a month ago -- looks like several references to the "BRUSHED_MOTORS" define were removed rather recently. 

It also seems that ESC/Motor features aren't sticking when set in the configurator.

mikeller, 2017-04-05:
This is caused by the configurator being out of synch with the firmware. There are a couple of changes that need to be ported from the cleanflight - configurator into the Betaflight - configurator to bring it into synch. This will happen shortly.

theseankelly, 2017-04-05:
Got it -- thanks for the quick reply! I'll wait on those changes.

theseankelly, 2017-04-07:
I experience the same behavior on the latest cleanflight-configurator -- is that expected?

theseankelly, 2017-04-13:
I found the cleanflight configurator that's linked to by the cleanflight 2.0 RC2 release and that doesn't cause this problem. Also, the copter works fine from my TX. I've now convinced myself that it's not a betaflight firmware problem :-)

closed 2017-04-13 by theseankelly

Catiadr, 2017-07-06:
I am trying to get this board to start without spinning the motors, but I haven't found the right setting.  The motors spin at 100% as soon as I plug the battery in.  I have tried many variations of the CLI command set motor_pwm_rate=xxxxx...Is anyone familiar with a fix? [https://www.banggood.com/Eachine-SKYkylin-F3-Brushed-Flight-Control-Board-Built-in-FRSKY-Compatible-8CH-Receiver-p-1100694.html?rmmds=detail-top-buytogether](url)

brucesdad13, 2017-07-06:
You're probably using 3.2 BF with the stable configurator. You need a matching configurator from development... https://github.com/betaflight/betaflight-configurator

Noctaro, 2017-07-06:
hmm, had the same problem with a matching configurator.