betaflight/betaflight #2829: VBAT no longer displayed on Taranis (v2.0.9) LCD after update from baseflight to betaflight

cooncat, 2017-04-06:
A couple weeks ago, I finally updated my oldest racing quad, a ZMR250 with Naze32, FrSky D4R-II from Baseflight (which originally came on the Naze32 Acro v5) to Betaflight v3.1.6.  After tuning, the ZMR250 has NEVER flown as well as it is now flying .. just great. However, I've lost the ability for VBAT to be recognized by my Taranis 9D+ (with v 2.0.9 fw). This WAS working under Baseflight. I believe the problem is that my Taranis allows for the VBAT source to be FAS, while Betaflight allows for FrSky telemetry VBAT to be sent out via VFAS.  The version of fw on my Taranis does NOT allow for "discover new sources" as the newer FW does. I'm hoping that there is an easy fix. (RSSI is still functioning and showing up on the Taranis telemetry screen).

mikeller, 2017-04-06:
Just to ask the obvious: Why don't you upgrade the firmware on your TX?

cooncat, 2017-04-06:
I did  upgrade the FW on my Taranis .. and had problems with several models. In the end, I thought it would be too much work to have to go thru every model and change settings to get things working again. So, I reverted back to v2.0.9.  When I did that, the newer rev. FW on the Taranis had somehow changed the data format so that my models were no longer recognized as valid in v2.0.9. So, in the end, I had to go thru and redo ALL my models when I went back to v2.0.9.  Bottom line, the Taranis works fine with v2.0.9, and I don't see any huge incentive to go thru the upgrade process again.  Thanks.

closed 2018-04-21 by DieHertz