betaflight/betaflight #2836: RC Rate and RC Expo on Pitch and Roll MERGED?

Holdox, 2017-04-07:
Why are they Merged? When did that happen? I want different settings on Pitch and Roll, and dont want to do it with super-rate. Have tried CLI, but could not figure out how to do it. Im sure there are people who want different roll and pitch rates.

mikeller, 2017-04-08:
They were always combined. Could it be that you are thinking of the (non - RC) rates, that are now called 'Super Rates'?

Holdox, 2017-04-08:
Really? You are probably right...no I was thinking about the RC's. The super-rate is not merged for some reason tho...

mikeller, 2017-04-08:
Yes, the super rates are individually adjustable, which makes sense.

Holdox, 2017-04-08:
I want to have slightly less sensitive rates on pitch than roll. I can set the rc rate lower and use srate on roll to do it probably. It is probably a hardware/software limitation that makes it impossible to have more controll over rates and curves? It would have been cool to draw own curves :-)

borisbstyle, 2017-04-08:
This actually has to do with support if guis and osds. They all support only one rc rate

closed 2018-04-21 by DieHertz