betaflight/betaflight #2837: set rc_interp = AUTO does not work correctly with Spektrum Satelites.

lunddahl, 2017-04-08:
rxRefreshRate is the problem, this debug is showing the problem, rxRefreshRate is switching between 3 and 21ms, the actual refresh rate is 22ms, but AUTO substracts 1ms...;-)

Screenshot:
[image]

Logfile:
[LOG00006.zip](https://github.com/betaflight/betaflight/files/907368/LOG00006.zip)

using:
rc_interp = manual + set rc_interp_int = 22 is working perfectly.

There is another problem, don't know if it's actually related, but in Europe, everyone who is using the new G2 EU version transmitters is hard limited to running DSMX (SPEKTRUM_2048) but ONLY at 22ms refresh rate.

```
    case SERIALRX_SPEKTRUM2048:
        // 11 bit frames
        spek_chan_shift = 3;
        spek_chan_mask = 0x07;
        spekHiRes = true;
        rxRuntimeConfig->channelCount = SPEKTRUM_2048_CHANNEL_COUNT;
        rxRuntimeConfig->rxRefreshRate = 11000;
        break;
    case SERIALRX_SPEKTRUM1024:
        // 10 bit frames
        spek_chan_shift = 2;
        spek_chan_mask = 0x03;
        spekHiRes = false;
        rxRuntimeConfig->channelCount = SPEKTRUM_1024_CHANNEL_COUNT;
        rxRuntimeConfig->rxRefreshRate = 22000;
        break;
```

We might need both SERIALRX_SPEKTRUM2048_11MS_FRAMERATE and SERIALRX_SPEKTRUM2048_22MS_FRAMERATE, the latter for us limited EU users.

There is NO way to put a non EU version software on a Spectrum TX, unfortunately.

Unfortunately i'm not nearly skilled enough to write C code, limited reading it is all i can do. ;-(

But if someone is interested in fixing this, i will be happy to debug.

borisbstyle, 2017-04-10:
maybe a preset value would be more suitable as a default rather than AUTO

lunddahl, 2017-04-10:
IMHO rxRefreshRate should be set when selecting the Serial Provider, RC Smoothing could then just be On or Off, or even more simple rc_interp_ch = NONE

To keep this from becoming a nightmare, there should be en option to use default for the provider or overriding it with you own value. 0ms could indicate use default for provider.

In this case i would know that my case is special, and i need to override the default 11ms with 22ms.

In the future we might see for example turbo modes in TX and RX systems, using the same protocols but with more updates. 

In case of non Serial RX, i dont know.

ledvinap, 2017-04-10:
@borisbstyle : IMO AUTO should calculate actual refresh rate (using timestamps and smoothing). From @lunddahl image it seems that calculation is wrong.

As for Spectrum values - even  11ms  rxRefreshRate  and 22ms update interval will help a lot - half of the interval will be ramp, second half will be constant. Large step in setpoint is removed, preventing step-induced oscillation.

lunddahl, 2017-04-10:
@ledvinap something is wrong for sure, the spektrum sat is sending data frames every 22ms, the 3ms detected is a detection error, and not what is happening on the serial line.

Spektrum sats sends a data frame every 11ms or 22ms, the assumptions that 10 bit packets is always 22ms and 11bit always 11ms is not right, Spektrum G2 TX EU models is in fact restricted to 11 bit packets and 22ms update period.

However detection is important, i think in case of other serial protocols, there is a chance for a lot more deviation in frame update speed from the RX.

lunddahl, 2017-04-10:
However i still think the optimal place to calculate refreshrate and/or set it manually should be in the serial provider not the smoothing code. I can see rxRefreshRate is also used with AntiGravityMode

```
 static void checkForThrottleErrorResetState(uint16_t rxRefreshRate) {
    static int index;
    static int16_t rcCommandThrottlePrevious[THROTTLE_BUFFER_MAX];
    const int rxRefreshRateMs = rxRefreshRate / 1000;
    const int indexMax = constrain(THROTTLE_DELTA_MS / rxRefreshRateMs, 1, THROTTLE_BUFFER_MAX);
    const int16_t throttleVelocityThreshold = (feature(FEATURE_3D)) ? currentProfile->pidProfile.itermThrottleThreshold / 2 : currentProfile->pidProfile.itermThrottleThreshold;

    rcCommandThrottlePrevious[index++] = rcCommand[THROTTLE];
    if (index >= indexMax)
        index = 0;

    const int16_t rcCommandSpeed = rcCommand[THROTTLE] - rcCommandThrottlePrevious[index];

    if(ABS(rcCommandSpeed) > throttleVelocityThreshold)
        pidSetItermAccelerator(currentProfile->pidProfile.itermAcceleratorGain);
    else
        pidSetItermAccelerator(1.0f);
}
```
I'm not very skilled at programming, so i have no clue if this refreshrate detection error can also gives problems with AntiGravityMode, and i haven't been able to test if it does.

I realize that changing such a fundamental thing could lead to all kind of new issues.

lunddahl, 2017-04-10:
If someone would like to try to fix this detection error, i will be happy to assist in testing and providing result logs.

borisbstyle, 2017-04-11:
@ledvinap actually the auto should work assumimg that rx is sending frames with same interval.
I have seen lately rx systems that can have variable refresh rate. That might be the case here

ledvinap, 2017-04-11:
@borisbstyle : Using 2point averaging filter will handle 2 different refresh intervals ...

ledvinap, 2017-04-11:
From first post screenshot, it seems that two packets arrive immediately (with little interpacket delay).
Can someone check this with logical probe? Maybe it is problem in BF code ..

borisbstyle, 2017-04-11:
@ledvinap well I don't want to create workarounds. Original assumption when this feature was created that ever rx has static refresh rate.
I might change default to use preset to be safer perhaps for now?

ledvinap, 2017-04-11:
@borisbstyle : That is reasonable assumption . But what if RX sends
`<pkt>0ms<pkt>22ms<pkt><0ms><pkt>22ms` ... ?

borisbstyle, 2017-04-11:
Than that is a problem.

Could do all kind of weird things like guarding against that and falling back to the highest detected or average of it, but would be nice to understand the behaviour.

ledvinap, 2017-04-11:
Logic analyzer trace is good starting point ... 
If RX really sends data this way, simple 2 point maximum filter will fix it. And won't affect normal receivers

lunddahl, 2017-04-12:
Unfortunately i dont have en Logic Analyser, but specifications for the protocol the satelites is using are now public:
https://www.spektrumrc.com/ProdInfo/Files/Remote%20Receiver%20Interfacing%20Rev%20A.pdf

> In normal operation, the Spektrum remotes issue a 16-byte data packet every 11ms or 22ms, depending
> upon the selected protocol. The packet is transmitted at 125000bps, 8 bits, No parity, 1 stop (8N1).
> For those UARTs which are not capable of that speed, they will also work at the more-standard
> 115200bps. When no data is received, the remote does not transmit anything. It will count missed
> frames, and the updated count will be sent after the next packet has been received but not until.

I think it's safe to assume that a fixed frame interval is used, there is a very big range of receivers, heli stabilizers and flight controllers supporting spektrum satellites, if they were to send frames like above debug indicate, many of those would probably malfunction as they are not even close to being as advanced at our FC with betaflight.

lunddahl, 2017-04-12:
If i recall correctly there was some code committed that fakes RSSI based on the fade count the satelite is sending, that code might have broken something, it recently, after RC smoothing was implemented.

But i can't find the commit....

ledvinap, 2017-04-12:
@lunddahl : From spectrum specification, only 7 channels fit in one frame. So it may be possible that two packets are sent when more channels are in use

lunddahl, 2017-04-13:
@ledvinap : No that is not the case, Spectrum transmitters are updating 4 channels in every frame, the rest of the channels is not transmitted in every frame, but in every other, some channels is even only transmitted in every third frame.

There was an update recently changing what four channels is transmitted in every frame, especially for multirotors, as the throttle channels was not one of them, and that is not good for multirotors, even Spektrum figured that out.

"For Multi-Rotor models, the high-speed (11ms) channels are THRO, AILE, ELEV, and RUDD.  For other model types AUX1 replaces THRO."

I have studied further into this 22ms update time for Spektrum EU radios, there is only two radios that was limited to 22ms in EU, the DX6G2 and the DX7G2, the DX6G2 is now replaced with DX6G3 that does support 11ms in EU and the DX7G2 is discontinued.

However, the bug still exists, but i can not test if it's actually there when running at 11ms update interval.

lunddahl, 2017-04-14:
@borisbstyle
@ledvinap 
I have studied the code quite some tonight, and i think we (mostly me) are interpreting the logs completely wrong.

```
void processRcCommand(void)
{
    static int16_t lastCommand[4] = { 0, 0, 0, 0 };
    static int16_t deltaRC[4] = { 0, 0, 0, 0 };
    static int16_t factor, rcInterpolationFactor;
    static uint16_t currentRxRefreshRate;
    const uint8_t interpolationChannels = rxConfig()->rcInterpolationChannels + 2;
    uint16_t rxRefreshRate;
    bool readyToCalculateRate = false;
    uint8_t readyToCalculateRateAxisCnt = 0;

    if (isRXDataNew) {
        currentRxRefreshRate = constrain(getTaskDeltaTime(TASK_RX),1000,20000);
        if (isAntiGravityModeActive()) {
            checkForThrottleErrorResetState(currentRxRefreshRate);
        }
    }

    if (rxConfig()->rcInterpolation || flightModeFlags) {
         // Set RC refresh rate for sampling and channels to filter
        switch(rxConfig()->rcInterpolation) {
            case(RC_SMOOTHING_AUTO):
                rxRefreshRate = currentRxRefreshRate + 1000; // Add slight overhead to prevent ramps
                break;
            case(RC_SMOOTHING_MANUAL):
                rxRefreshRate = 1000 * rxConfig()->rcInterpolationInterval;
                break;
            case(RC_SMOOTHING_OFF):
            case(RC_SMOOTHING_DEFAULT):
            default:
                rxRefreshRate = rxGetRefreshRate();
        }

        if (isRXDataNew) {
            rcInterpolationFactor = rxRefreshRate / targetPidLooptime + 1;

            if (debugMode == DEBUG_RC_INTERPOLATION) {
                for(int axis = 0; axis < 2; axis++) debug[axis] = rcCommand[axis];
                debug[3] = rxRefreshRate;
            }

            for (int channel=ROLL; channel < interpolationChannels; channel++) {
                deltaRC[channel] = rcCommand[channel] -  (lastCommand[channel] - deltaRC[channel] * factor / rcInterpolationFactor);
                lastCommand[channel] = rcCommand[channel];
            }

            factor = rcInterpolationFactor - 1;
        } else {
            factor--;
        }

        // Interpolate steps of rcCommand
        if (factor > 0) {
            for (int channel=ROLL; channel < interpolationChannels; channel++) {
                rcCommand[channel] = lastCommand[channel] - deltaRC[channel] * factor/rcInterpolationFactor;
                readyToCalculateRateAxisCnt = MAX(channel,FD_YAW); // throttle channel doesn't require rate calculation
                readyToCalculateRate = true;
            }
        } else {
            factor = 0;
        }
    } else {
        factor = 0; // reset factor in case of level modes flip flopping
    }

    if (readyToCalculateRate || isRXDataNew) {
        if (isRXDataNew)
            readyToCalculateRateAxisCnt = FD_YAW;

        for (int axis = 0; axis <= readyToCalculateRateAxisCnt; axis++)
            calculateSetpointRate(axis);

        // Scaling of AngleRate to camera angle (Mixing Roll and Yaw)
        if (rxConfig()->fpvCamAngleDegrees && IS_RC_MODE_ACTIVE(BOXFPVANGLEMIX) && !FLIGHT_MODE(HEADFREE_MODE))
            scaleRcCommandToFpvCamAngle();

        isRXDataNew = false;
    }
}
```
[image]

currentRxRefreshRate = constrain(getTaskDeltaTime(TASK_RX),1000,20000);
rxRefreshRate = currentRxRefreshRate + 1000; // Add slight overhead to prevent ramps
debug[3] = rxRefreshRate;

We are seeing an alternating RAW value of precisely 21000 or 2800~3200, that is actually the delta time of the RX_TASK + 1000 , it has nothing to do with how often RX data is received.

The RX_TASK is the only realtime task, and it is data driven OR with a fallback to 50Hz, when we see 21000 it is because we added 1000 to the deltatime of 20000us=20ms=50Hz, so we are just seeing the RX_TASK falling back for 1-3ms (that fit nicely with the time it takes to transmit a spektrum frame)

As you can see on the screenshots there is 22ms between when the task falls back, and that fits perfectly with the fact that exactly when the thread resumes from fallback rcCommand_raw[] is updated.

This leaves me with a warm feeling that i should remember that 20ms != 22ms AND a feeling that RC_SMOOTHING_AUTO might not be working as intended for anything where RX_TASK is data driven, like it is for all the serial rx protocols.

What are your inputs to this ?

borisbstyle, 2017-04-14:
The data rate doesnt have to be exactly the same for smoothing actually. A bit higher is fine too like 23 for 22ms to make the command transition smoother. Thats why 1ms is added

RC smoothing works actually really well as designed with data driven rx protocols faster than 50hz. 
What you are describing there is the problem with the scheduler maybe. When data driven protocol was slower than 50hz scheduler will execute the RX task again. RC_SMOOTHING_AUTO doesnt know weather it was triggered by real interrupt or timed based.

ledvinap, 2017-04-14:
IMO trivial fix is to decrease fallback frequency for digital receivers - it should not trigger anyway, so increasing it to 25ms (or even 40ms) should be pretty safe.

ledvinap, 2017-04-14:
RX code could use huge rewrite ... It was acceptable as experimental code, but now it's a bit headache ...

ledvinap, 2017-04-14:
@lunddahl :Good spot, I'll just reword your analysis a bit:

The problem is that in each cycle 50Hz fallback frequency triggers (20ms) and TASK_RX is activated.  Normal packet arrives ~2ms later (22ms update frequency) and fallback counter is restarted. getTaskDeltaTime returns difference between last two executions, so it returns ~2ms after data packet (delta between fallback frequency trigger and data trigger) and 20ms after fallback trigger (for 2 ms only). RC smoothing is calculated after receiving data packet when getTaskDeltaTime == 2ms, so there is short ramp.

So the problem happens only for data receivers with update rate lower than 50Hz/20ms. Extending fallback update frequency will fix it and remove edge case (but some other code may assume that TASK_RX frequency is at least 50Hz)

lunddahl, 2017-04-15:
@borisbstyle @ledvinap 
I have looked at a log today of a quad running SBUS at 11ms and that AUTO is actually getting the right data, when the TX is not sending packets with a spacing longer than 20ms.

So to recap, any serial protocol with a update interval longer than 20ms (19ms perhaps), will need to use RX  smoothing manual and set the interval manually.

```
set rc_interp = MANUAL
set rc_interp_int = 22
```

1. Only two Spektrum radios and only the EU versions is limited to 22ms, the DX6G2-EU and DX7G2-EU, and they are both discontinued/updated with newer models that do 11ms even in Europe, you would have to go very old to find a Spektrum radio that does not do 11ms.

2. It does not in general make sense to use transmitters with +20ms update frequency on today's multirotors.

3. Messing with changing the 50Hhz "limit" of current RX/Scheduler code might create other problems, IMHO the gain it not enough to justify it.

However AntiGravityModeActive seems to be using currentRxRefreshRate also, and that is not fixable by the user, furthermore a LOT of Spektrum TX'es is only updating the throttle channel every other packet, even when running 11ms, throttle will be 22ms, unless they use newest firmware AND QuadRacer Profile on the TX.

I have now clue if this is actually a problem ?