betaflight/betaflight #2838: No sticks input with IBUS on NAZE since build #1168

AirBreak69, 2017-04-08:
Using current build or release 3.1.7 on a good old NAZE with IBUS the stick inputs stuck at 1500.
Switches and VRs are still working.
Problem was introduced in build 1168.
Another problem with these and earlier builds: AUX2 is copied to AUX6, AUX6 does not work.

McGiverGim, 2017-04-08:
I use ibus shared with telemetry in uart2 and it is working for me in naze32. I had this problem but was solved I don't remember when.

unitware, 2017-04-08:
@McGiverGim are you on master or 3.1.7? 

https://github.com/betaflight/betaflight/issues/2637 was resolved on master like 16 days ago.

AirBreak69, 2017-04-08:
Just to give some hints:
I tried nearly all builds from 1163 to 1174 and the release 3.1.7. Build 1167 was the last working.
The effect may be very rare and isolated to may NAZE (Afro32).
IBUS in general is working. Roll, pitch and yaw stuck at 1500 but throttle is frozen to 885.
I also referenced this in cleanflight because the release 2.0.0 RC2 some 6 days ago was OK, but when the BF code was merged 2 days ago, cleanflight created the same problem on my particular setup.
I guess that something inside the anti gravity code makes the difference.

McGiverGim, 2017-04-08:
I don't know if your problem is this https://github.com/betaflight/betaflight/issues/2637
I've it, but now it's solved.