betaflight/betaflight #2839: OSD crosshairs not working

Arakon, 2017-04-08:
In both 3.1.6 and 3.1.7, the center crosshairs are not working. They're activated in the configurator, but simply not showing on screen.
[image]
[image]

lostcontrol, 2017-04-08:
Did you disable the accelerometer? As far as I remember, the crosshair will only be shown if the accelerometer is activated. I have [a commit for master in my repo](https://github.com/lostcontrol/betaflight/commit/91e80c9b95d44f210f8f7a0f5ea5dce8d1683944) to change this since I also don't see why the crosshair would require the accelerometer.

Arakon, 2017-04-08:
ACC is off, yes.. the crosshair is not dependant on it at all, so I agree, makes no sense. 
MWOSD enables crosshair even with ACC off.

mikeller, 2017-04-09:
@lostcontrol: Would you mind opening a PR for your fix?

lostcontrol, 2017-04-09:
@mikeller will rebase and open a PR

PR #2842 (2017-04-09): [CMS] Add crosshairs support
Add crosshairs support in CMS and also make the crosshairs visible even if the accelerometer is disabled. Should fix #2839
closed 2017-04-09 by mikeller