betaflight/betaflight #2840: DYS F4 OMNIBUSF4 target 3.1.7 SBUS doesn't work, 3.1.6 SBUS ok

sevet, 2017-04-08:
Hi,

Just got the new DYS F4 it uses the OMNIBUSF4 target, for some reason the SBUS is broken on 3.1.7 and is working on 3.1.6 and also on 3.1.5

Other issue FIY that UART3 tx and rx on the documentation is swapped... also it is un inverted.

jflyper, 2017-04-09:
SBUS is working on the original OMNIBUSF4 with v3.1.7.

Need a hardware for further investigation.
And about the document, is that BF origin?

dronejunkie, 2017-04-13:
Project blue falcon did a review on this fc. He said it is broken in 3.1.7

https://youtu.be/q-569po-ipU

jflyper, 2017-04-13:
@dronejunkie Sorry, some one with an actual hardware has to trouble shoot it; like I said above, it is working on the '*reference*' board OMNIBUS F4 AIO.

[image]

RobotGod, 2017-04-14:
I experienced same problem with 3.1.7 and OMNIBUSF4SD, had to revert back to 3.1.6 due to sbus on uart1 does not seem to work.

jflyper, 2017-04-16:
@RobotGod Can't reproduce. It's working for OMNIBUS F4 V2 and V3, F4 Pro V3.

[image]

jflyper, 2017-04-16:
There must be something that does not develop when tested with a bare board + a receiver.

Those of you having problem, can you write out what you have connected?

RobotGod, 2017-04-16:
OMNIBUSF4SD + XSR + RacerStar 4in1 w/Dshot600.  I went back and forth from 3.1.6 to 3.1.7 a few times to verify.  It didn't work at all with 3.1.7 100% of the time.

jflyper, 2017-04-16:
@RobotGod Nothing special... What if you change to OS or even PWM?

sevet, 2017-04-16:
I used the DYS F4 new board with PDB (sbus didnt work on 3.1.7)

jflyper, 2017-04-17:
@sevet No new info. Have you tested with ONLY the receiver?

dronejunkie, 2017-04-17:
AFAIK the issue is the inverter​ hardware is not working on the FC. If you have a xsr receiver u can try uninverting the signal on the receiver. https://youtu.be/gWg9R7OLe5c

jflyper, 2017-04-17:
@dronejunkie It's more than that. If it was a simple problem like "inverter not working", 3.1.6 does not work either.

Can someone investigate the inverter circuit on this FC, if it is transistor based or gated or something else?

dronejunkie, 2017-04-17:
What I meant was in 3.17 the inverter setting is not working on this
particular fc.

dronejunkie, 2017-04-17:
Correction checked the betaflight notes sbus invert is only for F3 board.
Me bad.

fncisco, 2017-04-27:
I didn't see on deep, but someone change the inverter pinout on target, recently the target have many changes. one of this was the default serial port and inverter pin. we must see what is the real pin and fix it.

jflyper, 2017-04-27:
@fncisco 
That is not true. For 3.1.7, the inverter pin def was only changed for OMNIBUSF4SD target; OMNIBUSF4 remained the same. And there is no such thing as default UART.

However, the inverter pin def is queued (planned) for deletion, because it is not used by the original OMNIBUS F4 AIO.

digaus, 2017-05-02:
The Flytower F4 has the same problem (OmnibusF4SD target). 3.1.6 sBus works fine but on 3.1.7 it does not work anymore.

isilcala, 2017-05-11:
I got a DYS F4 v1.1, has 6 more pins for RSSI & 3.3v. It seems better if we create a new target for this FC?

pango, 2017-05-11:
I have the DYS F4 and it also has the SBUS Bug with BF 3.1.7 How can I help to solve this issue?

mikeller, 2017-05-11:
@pango: Yes. Complain to dys support and ask them to provide hardware information so their hardware can be properly supported.

dronejunkie, 2017-05-12:
Totally agreed, this is not good for their product. I believe the Omnibus
FC also has the same issue.  It is in the vendor's interest to fix this
ASAP.

liquidacid, 2017-05-14:
On 3.1.7 PC0 (Pin8) to the inverter is about 0.6v (3.3v on 3.1.6), missing pullup on PC0? I will probably test this within a few hours.

mikeller, 2017-05-14:
@liquidacid: Is this using the OMNIBUSF4 or the OMNIBUSF4SD target? UART1 inverter pin was shifted from PC0 to PC8 for OMNIBUSF4SD in 3.1.7.

isilcala, 2017-05-15:
@mikeller  OMNIBUSF4 target.

jflyper, 2017-05-15:
@here 
Those of you having issues with SBUS inversion, take a look at the note on configurable inverter in OMNIBUS F4 board wiki.

(1) Get the nightly build.

(2) Try this
```resource inverter 1 c0```

pango, 2017-05-15:
@mikeller Done.  I did complain to dys in the main dys F4 rcgroups Thread.  ;)

DieHertz, 2017-05-15:
https://www.rcgroups.com/forums/showpost.php?p=37523147&postcount=300
Someone mentions that it's indeed inversion thingie.
But how could that happen if OMNIBUSF4 has not changed, only OMNIBUSF4SD has?

mikeller, 2017-05-16:
@DieHertz: There are a lot of clones of this board so it's possible that some of the clones use a different pin, and the reports just make it look like it ever worked with the OMNIBUSF4 target.

isilcala, 2017-05-16:
@mikeller not sure about clones but I can confirm that I have a legit v1.1 board from DYS, it works with 3.1.6 OMNIBUSF4 target but not 3.1.7.

mikeller, 2017-05-16:
@isilcala: :-) The dys board is a clone, the original OMNIBUS  series was developed by @nathantsoi with myairbot (I believe).

But nonwithstanding, if your board is using the OMNIBUSF4 target, the inverter for UART1 should either work on 3.1.6 and 3.1.7, or not work on both. Are you sure that the 3.1.6 that you tested with is vanilla 3.1.6, or is it a (potentially patched) version that came preinstalled?

DieHertz, 2017-05-16:
@mikeleer That's a good explanation! They indeed come pre-flashed with 3.1.6.

DieHertz, 2017-05-16:
Someone with a multimeter, please find which MCU pin is connected to the inverter control input. It could help us solve this issue easily.

pango, 2017-05-16:
@mikeller my dys board had 3.1.6 preinstalled. I installed 3.1.7 and SBUS did not work. After that I installed 3.1.6 via BF Firmware Flasher with a online download and sbus is working again. Details of 3.1.6 Version are:
2017-05-16 @ 12:15:27 -- MultiWii API version received - 1.31.0
2017-05-16 @ 12:15:27 -- Flight controller info, identifier: BTFL, version: 3.1.6
2017-05-16 @ 12:15:27 -- Running firmware released on: Feb 21 2017 14:33:46
2017-05-16 @ 12:15:27 -- Board: OBF4, version: 0

DieHertz, 2017-05-16:
Please try whether SBUS works with this custom build.

[betaflight_3.1.7_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/1004079/betaflight_3.1.7_OMNIBUSF4.hex.zip)

DieHertz, 2017-05-16:
Please try this custom HEX.
[betaflight_3.1.7_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/1004104/betaflight_3.1.7_OMNIBUSF4.hex.zip)

pango, 2017-05-16:
It does something now, but not as intended I would say. 
When I raise only the Throttle stick, the three channels Throttle yaw and AUX1 are freaking out.
The same channels go crazy when I move Roll or Pitch.
But on the actual Roll and Pitch Channels is no input at all.

mikeller, 2017-05-16:
@pango: That's when receiving from an SBus RX? The symptoms sound like what you typically get when having SBus and SUMD mixed up, can you do a quick check to make sure your RX type is correct?

DieHertz, 2017-05-16:
Don't think it's related, SBUS either works or not. If it started working, the freaking out issue cause lies somewhere else.

DieHertz, 2017-05-16:
Since 3.1.7 is a history anyway, out of my curiosity please give current master a try as well.
[betaflight_3.2.0_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/1004269/betaflight_3.2.0_OMNIBUSF4.hex.zip)

pango, 2017-05-16:
3.2.0 is working.
and after a retest also 3.1.7 is working fine now.
@mikeller was correct, sorry, that was my fault as I didn't switch sbus provider from spektrum to sbus while testing.
Thank you very much.
Great Job!!!

DieHertz, 2017-05-16:
Great, thanks for testing.

closed 2017-05-16 by DieHertz

mikeller, 2017-05-17:
@DieHertz: I never said it wasn't user error. :-P There are little to no provisions for data validation built into SBus, so anu data stream that isn't SBus will contain parts that evaluate as valid SBus frames, with the consequences described above.

isilcala, 2017-05-17:
@mikeller thank you ;) yeah I understand the DYS FC is a clone of Omnibus, I thought you were saying a clone of the DYS :P 
so now it seems when the 3.2 is officially released this issue will be gone right?

mikeller, 2017-05-17:
The issue will be fixed in the next release. (Planned to be 3.2, unless a critical issue needs to be urgently fixed before 3.2.)

DieHertz, 2017-05-17:
@mikeller I can't recall opposing you on this subject :-)

liquidacid, 2017-05-22:
For now, those who want to have it working on 3.1.7, here you can find a special compiled version of the firmware: https://quadcoptersenzo.nl/index.php?route=module/mmos_attachmanager/getfile&product_attach_file_id=269

dronejunkie, 2017-05-22:
> index.php?route=module/mmos_attachmanager/getfile&product_
> attach_file_id=269

Will there be a new target for the dys f4 in 3.2? Otherwise it will break
again.

DieHertz, 2017-05-22:
@liquidacid A 3.1.7 compiled from a tag is 6 comments up from there, why another link?
@dronejunkie there won't as it's too similar to OBF4, nothing stands out and another target will just add maintenance burden

liquidacid, 2017-05-22:
@DieHertz  Im sorry, i didn't see that one. I'll remove it now :)

gregurbo, 2017-05-24:
@DieHertz so is the link for the working 3.1.7 just here in this thread or is it ok to use 3.1.7 from the configurator ?

DieHertz, 2017-05-24:
There's no way to update existing 3.1.7 entry in configurator AFAIK.

pkendall64, 2017-05-26:
I believe that this bug was introduced in commit 6aee42da
I the name of the inverter defines should be for USART* not UART* in the 3.1 codebase as of the 3.2 codebase (master) the names have changed. I think this bug is also in the KROOZX and BEEROTORF4 targets.

titanrw, 2017-05-28:
Most of this bug report is mentioning clones.  Just wanted to be clear, I have an original non clone omnibus f4 pro purchased direct from myairbot.  SBUS also broke with 3.1.7.  Worked fine on 3.1.6.  Haven't tried the modified 3.1.7 listed above as I'll just wait for 3.2 and run with that.

rodanone, 2017-06-22:
I wanted to add something to the FLYTOWER F4  OMNIBUSF4SD comment above. I was able to get past the SW version issue with the inverter in hardware by locating the LVC86 little logic device SC-70-5 package (Top marking 'CHF') on the board near the SBUS connector.  

I confirmed that between version 3.1.3 and 3.1.6 the invert signal on pin 1 of this device was changed from a '1' to a '0'. Subsequently, the SBUS communication stopped working. I don't agree with the above that it still worked in 3.1.6. I found the failure started after 3.1.3.

I lifted pin 1 of the XOR device from the board and directly jumpered the A pin (1) to the VCC pin (5).

After completing this simple modification the polarity of the SBUS is correctly received by the FC and so far the software and controller are functioning properly.

I hope that this note helps people. I have six vehicles based on these Flytowers and I really wanted to use the latest firmware.

If the software is corrected for this board later I will remove the logic device and simply replace it.

jflyper, 2017-06-22:
@rodanone 
Take a look at OMNIBUS F4 board wiki.