betaflight/betaflight #2846: TBS Unify Pro Smartaudio v1 in pitmode?

lasharor, 2017-04-09:
I'm flying with an SPracingF3 and TBS Unify Pro with Smartaudio V1. Have softserial port remapped to Motor output 7 since regular softserial ports would not work. 

Configuration using the lua script works as expected. However, I'm experiencing a lack of signal after turning the quad on. The video transmitter will only start transmitting when the channel is changed. No transmission after powering on the device.

This is strange behaviour, since the TBS Unify pro does not have a pit mode and should just transmit after powering on. Unclear if this "feature" is by design or this is a bug.

jflyper, 2017-04-10:
@lasharor The issue has been raised in a RCG thread, and I have a branch that fix it.
It's going to be great if you can test it.

[betaflight_3.1.7_SPRACINGF3.hex.zip](https://github.com/betaflight/betaflight/files/909139/betaflight_3.1.7_SPRACINGF3.hex.zip)

Chefhandwerker, 2017-04-21:
I have the same issue on a Omnibus (F3). Do you have a Fix for it?

jflyper, 2017-04-22:
@Chefhandwerker You can try the hex right above your comment.

Chefhandwerker, 2017-04-22:
@jflyper  I have test it. But the hex do not work on the omnibus

jflyper, 2017-04-23:
@Chefhandwerker Sorry, I'll make one for OMNIBUS, too.

Chefhandwerker, 2017-04-23:
@jflyper , perfekt. I use a Omnibus (F3) und a Unify pro (5V) V1. The Firmware I use . I the 3.1.7 Omnibus

[betaflight_3.1.7_OMNIBUS.zip](https://github.com/betaflight/betaflight/files/949675/betaflight_3.1.7_OMNIBUS.zip)

jflyper, 2017-04-23:
@Chefhandwerker 
[betaflight_3.1.7_OMNIBUS_bfdev-smartaudiov1-fix.hex.zip](https://github.com/betaflight/betaflight/files/949731/betaflight_3.1.7_OMNIBUS_bfdev-smartaudiov1-fix.hex.zip)

Chefhandwerker, 2017-04-23:
@jflyper ,Thank you. Perfect work!!

philippamlong, 2017-04-24:
@jflyper 
Hi, did you merge the fix to master already? If not, is your branch accessible in github? 

Thank you,
Philipp

jflyper, 2017-04-25:
@philippamlong 
Haven't submitted a PR yet (need few more testers).

This is the branch.
https://github.com/jflyper/cleanflight/tree/bfdev-smartaudiov1-fix

philippamlong, 2017-04-25:
@jflyper 
Good news, got my the electronics for my new drone on the testbench today.

Testsetup: omnibus f3, old unify pro from the first batch

Test 1: Hardware UART 2 -> failed, no VTX found
Test 2: Softserial assigned to B6 -> VTX found, but sends only after changing the channel
Test 3: Softserial assigned to B6 and your 3.7.1 build -> VTX found and sends directly!

Tomorrow i could test the your SPF3 build against a new smartport v2 unify hv on soft serial, just to be sure it will not affect it. Should i, or did you already test that?

jflyper, 2017-04-26:
To those who want to test the V1 fix on BETAFLIGHTF3:
[betaflight_3.1.7_BETAFLIGHTF3.hex.zip](https://github.com/betaflight/betaflight/files/957281/betaflight_3.1.7_BETAFLIGHTF3.hex.zip)

Stavr0s91i, 2017-05-14:
Hi i have the same issue on OmnibusF4 can i get a .hex of the latest 3.1.7 with the fix ?

Thanks :)

jflyper, 2017-05-15:
@Stavr0s91i The firmware hex was for testing the fix. The fix is already included in the master dev branch, and will be a part of the 3.2 release (probably be available in a month or so).

zschiffelbein, 2017-06-01:
I am experiencing the exact same issue. Running Betaflight 3.1.6 on an SPRacingF3 with a TBS Unify Pro V1. SmartAudio is wired to SoftSerial1. Will test out the hex file tonight. Thank you!

wolverin3, 2017-06-03:
@jflyper : I tested out above hex for BF F3 FC but it does not work for my TBS Unify Pro 5.8G with Smartaudio v1. 

I wired up the Smartaudio wire to pin B07 and remapped it using below command to SoftSerial 1.

resource serial_tx 11 b07

Please advise. Thank you.

wolverin3, 2017-06-03:
Pin b07 aka PPM pin on BF F3 FC.

zschiffelbein, 2017-06-05:
Confirmed that the above hex file resolves the issue on my setup. Thank you.

jflyper, 2017-06-11:
@wolverin3 What does the line for B07 says in "resource list" output?

wolverin3, 2017-06-11:
Hi @jflyper:

# resource list
IO
A00: FREE 
A01: OSD_CS 
A02: FREE 
A03: SERIAL_RX 2
A04: ADC_BATT 
A05: ADC_CURR 
A06: MOTOR 1
A07: MOTOR 2
A08: LED_STRIP 
A09: SERIAL_TX 1
A10: FREE 
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: MPU_CS 
B00: FREE 
B01: FREE 
B02: FREE 
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: FREE 
B07: FREE 
B08: MOTOR 3
B09: MOTOR 4
B10: FREE 
B11: FREE 
B12: SPI_CS 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: FREE 
C15: BEEPER 
F00: FREE 
F01: FREE 
F03: FREE 
F04: FREE 

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: LED_STRIP
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: MOTOR 1
DMA1 Channel 7: MOTOR 4
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: MOTOR 2
DMA2 Channel 4: FREE
DMA2 Channel 5: MOTOR 3

wolverin3, 2017-06-11:
@jflyper : I'm using B07 based on the info from this URL:

https://github.com/betaflight/betaflight/wiki/Single-Wire-Software-Serial

wolverin3, 2017-06-11:
# diff
# Betaflight / BETAFLIGHTF3 3.1.7 Apr  3 2017 / 21:43:25 (e1c4b5c)
name Wolverine
resource PPM 1 NONE
resource SERIAL_TX 11 B07

feature -BLACKBOX
feature -SDCARD
feature SOFTSERIAL
feature LED_STRIP
feature AIRMODE
serial 0 32 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
led 0 4,2::COW:3
led 1 5,2::COW:3
led 2 6,2::COW:3
led 3 7,2::COW:3
led 4 8,2::COW:3
led 5 9,2::COW:3
led 6 10,2::COW:3
led 7 11,2::COW:3
aux 0 0 0 1300 2100
aux 1 4 0 1300 2100
aux 2 13 2 1700 2100
set rssi_channel = 8
set fpv_mix_degrees = 30
set motor_pwm_protocol = DSHOT300
set acc_hardware = NONE
set pid_process_denom = 1
set osd_units = METRIC
set osd_rssi_alarm = 45
set osd_cap_alarm = 1200
set osd_time_alarm = 3
set osd_vbat_pos = 2472
set osd_rssi_pos = 2490
set osd_flytimer_pos = 2518
set osd_ontimer_pos = 374
set osd_throttle_pos = 2497
set osd_horizon_pos = 200
set osd_current_pos = 2465
set osd_mah_drawn_pos = 2480
set osd_craft_name_pos = 2504
profile 0

set d_lowpass_type = PT1
set p_pitch = 63
set i_pitch = 45
set d_pitch = 25
set p_roll = 49
set i_roll = 35
set d_roll = 20
rateprofile 0
rateprofile 0

set rc_rate = 150
set rc_rate_yaw = 130
set rc_expo = 15

jflyper, 2017-06-11:
Your config look fine...

Did you try other possibilities (pins)?

wolverin3, 2017-06-11:
I didn't try other pin. 

Do I need to use the hex provided by you?

It should be working for Smartaudio v1 right?

jflyper, 2017-06-16:
@wolverin3 You can use recent nightly builds.

wolverin3, 2017-06-16:
The thing is, I cannot even arm my quad using Betaflight 3.2 nigthly build. Have you tested this Smartaudio v1 on another BF F3 FC? It is really working on it? Thank you.

jflyper, 2017-06-18:
@wolverin3 I don't have the BFF3 here, but 3.2 nightly build should be arm-able.
(1) Did you do the full chip erase?
(2) How did you configure after flashing?

wolverin3, 2017-07-21:
I can arm ny quad now with BF 3.2 nightly build but my SmartAudio v1 still does not work.

jflyper, 2017-07-26:
@wolverin3 What is your Unify?

wolverin3, 2017-07-26:
Tbs unify 5g8 with smartaudio v1. The earliest version.

jflyper, 2017-07-26:
@wolverin3 Product URL please?

wolverin3, 2017-07-26:
http://team-blacksheep.com/products/prod:unify_pro_hv_sma

jflyper, 2017-07-26:
I think that device speaks V2...

And should be compatible with the current firmware...

wolverin3, 2017-07-27:
Mine is the earlier version, serial no below 1,000... it is v1.

jflyper, 2017-07-27:
@wolverin3 I'm sorry, we have to scope your signal to do something beyond this point.

jflyper, 2017-07-27:
@wolverin3 I've tested the functionality of SmartPort (V1) over software serial on STM32F303CC PB7 using an OMNIBUSF3 and a bread board based dev env, and they both worked without a problem, suggesting your problem is not a software related one.

These are the relevant lines from cli "diff".
```
resource SERIAL_TX 11 B07
feature SOFTSERIAL
serial 30 2048 115200 57600 0 115200
```
And these are required for me to manipulate CMS
```
feature -RX_PPM
feature RX_SERIAL
serial 2 64 115200 57600 0 115200
set serialrx_provider = SBUS
```

By entering these 7 lines of config followed by full-chip erase or `defaults`, I can make the SmartAudio work on the two environments over and over.

I suspect one or more of the following happening.

(1) Your wiring is incorrect (connected to somewhere other than PB7).
(2) Your wiring is dud (cold joints, unstable joint, internally cut wire, etc.)
(3) Your PB7 is damaged.
(4) Your Unify is not talking valid SmartAudio protocol.

jflyper, 2017-07-27:
@wolverin3 If you vrx is capable of receiving audio and you have something to monitor the audio output, it would add a bit of information to your investigation work.

You should hear pulsating noise on audio.
Ref:
https://www.youtube.com/watch?v=YuOZ0XNma1A

wolverin3, 2017-07-27:
Noted with thanks.

Btw, does this work in BF 3.1.7 too or just 3.2 only?

jflyper, 2017-07-27:
Better to do it with 3.2.

3.1.x does not support V1; it sort of works (like becoming online), but that's it.

wolverin3, 2017-07-28:
Yes, like you said, it seems my soldering joint is the problem, looks fine visually, wasted so much of my time.

Btw, although I can get it online now and change the frequency, but it seems not to save it after reboot. It is always using unknown freq (unable to scan) after rebooted and I need to manually change the freq again if I want to fly the quad. Is this the prob that you mentioned when using BF3.1.7? 

Thank you very much!

jflyper, 2017-07-28:
> Btw, although I can get it online now and change the frequency, but it seems not to save it after reboot. It is always using unknown freq (unable to scan) after rebooted and I need to manually change the freq again if I want to fly the quad. Is this the prob that you mentioned when using BF3.1.7?

Exactly.

V2 supports pit mode, and there is a command code that investigate frequency set for the pit mode. The 3.1.7 driver use this command code to obtain the frequency during initialization. Unfortunately, this command code is interpreted as set frequency command by V1 devices, that throws V1 devices into unknown frequency during the initialization.

https://github.com/betaflight/betaflight/blob/master/src/main/io/vtx_smartaudio.c#L718-L724
```
 718     case 1:
 719         // Don't send SA_FREQ_GETPIT to V1 device; it act as plain SA_CMD_SET_FREQ,
 720         // and put the device into user frequency mode with uninitialized freq.
 721         if (saDevice.version == 2)
 722             saSetFreq(SA_FREQ_GETPIT);
 723         ++initPhase;
 724         break;
```

wolverin3, 2017-07-28:
Done, flashed BF 3.2 nightly and it is working perfectly.

You are the man @jflyper!

Thank you so much for your kind reply and for your expertees.

Cheers bro!!

jflyper, 2017-07-28:
@wolverin3 That is good to hear. Happy props now! :)

doobran, 2017-07-30:
Hi all! I have just got my V1 Unify working as well thanks to these comments! The only problem I am having now is that when even you repower the quad and the BFF3 OSD loads it puts the unify into a mode where I cannot see anything on any channel. I have PIDflight and when it reads the settings it says the VTX is still on the channel I left it. To get the image back I have change the channel to another channel other than I had it on. You cannot leave it on the same channel on the app and press set... Nothing happens. This is a pain in the ass, especially if I am changing settings and I move my sticks to save and reset on the OSD... it instantly changes the frequency to nothing again. FYI OP model is set to FREE, PIT Fmode is PIR, (POR freq is 0 - it won't let me set it to anything else).

jflyper, 2017-07-30:
@doobran 

If you had that much time for typing, you could read few comments up.

closed 2017-07-30 by jflyper

doobran, 2017-07-30:
Wow, no need to be rude. I found the comment, I missed it before. Ill try that solution now.

jflyper, 2017-07-30:
@doobran Sorry, I wanted to be funny 😅