betaflight/betaflight #2848: BF 3.1.6 with SP F3 download blackbox log freezes #2799

attilafustos, 2017-04-09:
This is issue #2799 reopened.
Today I filled my SP F3 blackbox with data and I cannot download.
I disconnected the OSD but it does not help.

mikeller, 2017-04-10:
Have you tried a different / shorter cable and / or downloading from a different PC?

attilafustos, 2017-04-10:
I have tried 3 USB cables. Will try on a different PC also but I tried with BF 2.9.0 and it worked.

mikeller, 2017-04-10:
Betaflight 2.9.0 used smaller block sizes for dataflash download, causing the speed to be considerably slower. The new larger block size / faster speed works fine for most users' hardware combinations, but seems to be causing trouble with a few setups. We will look into making the block size adjustable for these cases,

closed 2018-04-21 by DieHertz