betaflight/betaflight #2850: Access SD via USB

Gaz-B, 2017-04-09:
Is it possible to read the BB SD via the USB I have a frame with limited access . The new Kiwi V2 would be perfect is this was possible John lemon of Flying lemons Believes this would need to be done in betaflight and the hardware on his board would be capable.

mikeller, 2017-04-10:
At the speed that is feasible with the currently used MSP protocol ( a couple of 100 kB / second), reading logs of any size sounds like an arduous task.

As a suggestion, it should be possible to have the flight controller boot into an alternative mode where it emulates an SD Card reader (and code for this probably already exists).

rav-rav, 2017-04-10:
I would love to see that, too, as I always forget the card in my sd card reader and go fly without it.

Gaz-B, 2017-04-10:
Booting in a different mode is exactly what I expected it to need maybe A CLI command similar to typing DFU instead of getting to the button what Godsend that is.

lunddahl, 2017-04-10:
Is it possible to load and run such code from the SD card itself, if so reading logs from SD card is only the beginning, updating (or dumping) firmware or configurations could be another use case.

nathantsoi, 2017-04-11:
maybe something like this, if somebody wants to implement it? https://community.st.com/thread/20414#comment-43657 (per: https://community.st.com/thread/36976-need-usb-mass-storage-device-example-code-for-stm32f4-discovery-board)

mikeller, 2017-04-11:
Considering how common the 'read data from attached SD Card' use case is, I would expect that there are probably plenty of good examples of how to do it out there, like the one you mention, @nathantsoi.

One thing I am still humming and harring about is on how to invoke this. Doing it by MSP / configurator is obviously one way (that should be done), but the usability isn't great, particularly since it's asymmetrical, i.e. once the flight controller is in 'SD Card reader' mode, it cannot be taken out of it by MSP / configurator, since it won't be recognised as a flight controller in that mode. My thoughts go towards adding a pull up / down button to one of the pins (potentially a motor pin), and checking that pin at boot time. If a button press is detected, the flight controller would then initialise in 'reader mode'.

Thoughts, everybody?

Gaz-B, 2017-04-11:
I don't see the issue with using the configurator to enter SD the mode as long as when power cycled it starts back in FC mode.

DanNixon, 2017-04-11:
@mikeller One way to exit mass storage mode could be using a "flag" file that once deleted will return to FC mode.

mikeller, 2017-04-12:
@gaz-b: It's not an issue, but having to use the configurator restricts the usefulness of the feature. Imagine wanting to download logs onto a device that doesn't have configurator installed, doesn't even support it, or can't even deal with USB serial devices (think iPhone).
@dannixon : Yes that's one way to do it. I would be going for just power cycling the board without pushing the 'card reader' button, much like boot mode.

nathantsoi, 2017-04-12:
i was just saying we should use mass storage instead of serial. im no usb expert, but i think we just want a composite device w/ cdc + msc

unitware, 2017-04-19:
could a mass storage device be used with a special /dev/msp device file that is the serial port connection? and /dev/esc1, /dev/gps..?

mikeller, 2017-05-22:
@unitware: Not really, since unix style device files are just data structures that hold numeric device ids, which in turn are used by the operating system to look up how to access a given device.

unitware, 2017-05-22:
Yes, I was more thinking of a custom handling of certain "device files", that the FC presented itself as mass storage dev but the firmware intercepts file system calls to these specific "files" and does MSP / ESC magic with them.

The benefit would be higher throughput, easy access to BB data files and easy access to passthrough ports. Just an idéa that is not that well refined.

mikeller, 2017-05-22:
@unitware: Ok, I get you. I've seen some GPS devices that report as two independent USB devices, one for the SD card, and one that is a virtual file system on the device, with a config 'file' that can be read and written. That would certainly be a possibility. In a first step I think it would make sense to implement an 'USB SD card reader mode' only, because that is what's missing, the existing 'normal mode' works fine for everything else.

Dutchy-79, 2017-07-14:
Sure hope this will be in a new update, My card is unreachable in my hexacopter setup. I will have to take it apart te get to the card.

^^BTW, Frsky Taranis also shows up as 2 drives

mikeller, 2017-08-16:
@JonnyPhenomenon: There definitely is a lot of interest in getting this added. All we need now is a developer with time on their hands to write the code for it. ;-) (Any help needed to get this integrated into Betaflight is available...)

Dutchy-79, 2017-08-16:
Yes would be perfect for tight builds, and if it's implemented board designers can work on better locking systems for the card, no more lost cards on the field.

NatroNx, 2017-12-27:
I would love to see this as well.  More and more boards come with SD Card option, but most are hard to Access.

Matthias7Song, 2017-12-27:
Raceflight does that. Wen Flight Log Download Mode is enabled followed by FC reboot, flash RAM is mounted as an external storage device. Super fast log download or copy.

DieHertz, 2017-12-27:
Thanks for letting us know

DieHertz, 2018-04-21:
Implemented by #5443 thanks to @chrish

closed 2018-04-21 by DieHertz