betaflight/betaflight #2851: Softserial unstable on X-Racer F303 3.1

lassebm, 2017-04-09:
(Issue created after discussion with @jflyper on Slack.)

SmartPort telemetry is unstable through softserial when a battery is connected to the quad.

With SmartPort connected to CH6 (SOFTSERIAL1), the frequency of FC telemetry will gradually slow down until no FC telemetry is received. After typically 10 seconds, FC telemetry is again received and the pattern repeated.

With SmartPort connected to CH1 (remapped with `resource serial_tx 11 a0`), **no** FC telemetry is received when a battery is connected.

lassebm, 2017-06-12:
@jflyper Any chance you'll find time to look into this issue? If not, I might just buy another FC with built-in OSD to free up a hardware serial. Thanks.

jflyper, 2017-06-13:
@lassebm I can't reproduce the issue; I don't have the X-Racer F303 test board.

Let me clarify few things (Help me recall these)
- Does the issue develop if powered ONLY by USB?
- Does it always work stably if hardware UART is used?

lassebm, 2017-06-19:
@jflyper After further investigation it seems to be caused by an intermittent disconnect in the PCB layers. When the board is warmed up, the issue shows up and a light touch will make the issue go away. Thanks for your support!

jflyper, 2017-06-20:
@lassebm Sorry to hear that...

closed 2017-06-20 by lassebm