betaflight/betaflight #2852: BF 3.1.7 Gyro/Pitch issue

chime13, 2017-04-09:
I'll be flying along happily in a forward direction and I decide to punch full throttle - immediately if not sooner, the quad will do a backwards flip/stall and crash.
If I'm hovering and then punch it, it does shoot straight up - completely level (for the most part)

After updating all my FCs to 3.1.7, This now happens to all 3 of my quads. (8.5mm brushed builds)
FCs = Microscisky, Naze and SPRACINGF3_EVO version 3.1.7

Was not able to tune this issue away.
Solution: Downgraded to version 3.1.6 and no longer have this issue on any of my quads

Video of Quad 1 having issue
https://youtu.be/D6514uEMEJE

Video of Quad 2 having same issue
https://youtu.be/KwRghzmQuhI

borisbstyle, 2017-04-10:
Can you post your "diff" output as well.

It looks to some kind of level mode kicking in like it is a failsafe happening?

chime13, 2017-04-10:
Hey Boris - Here's the diff from my 3.1.6 which was identical in 3.17

I got no impression that it was a failsafe issue and I only fly Acro. I don't use Angle or Level mode. I have been playing with anti-gravity gain and did set it back to 1.000 when this issue cropped up. I use 4.000 on all my builds now without any issues.

# diff
# Betaflight / SPRACINGF3EVO 3.1.6 Feb 21 2017 / 14:44:42 (1fd502c)
name OrangeF3EVO

feature -RX_PPM
feature -FAILSAFE
feature -TELEMETRY
feature -BLACKBOX
feature -TRANSPONDER
feature -SDCARD
feature RX_SERIAL
feature MOTOR_STOP
map TAER1234
serial 2 64 115200 57600 0 115200
adjrange 0 3 0 1675 2100 12 0
rxrange 0 1050 1947
rxrange 1 1051 1947
rxrange 2 1051 1947
rxrange 3 1051 1947
set auto_disarm_delay = 3
set serialrx_provider = SPEK2048
set gyro_sync_denom = 4
set yaw_motor_direction = -1
set acc_hardware = NONE
set baro_cf_vel =  0.985
set pid_process_denom = 2
profile 0

set vbat_pid_gain = ON
set anti_gravity_gain =  4.000
set setpoint_relax_ratio = 10
set d_setpoint_weight = 85
set p_pitch = 95
set i_pitch = 35
set d_pitch = 55
set p_roll = 95
set i_roll = 35
set d_roll = 55
set p_yaw = 120
set i_yaw = 90
rateprofile 0
rateprofile 0

set rc_rate = 105
set rc_rate_yaw = 105
set rc_expo = 40
set rc_yaw_expo = 40
set thr_mid = 40
set thr_expo = 60

#

Arakon, 2017-04-10:
So anti_gravity_gain at 4 fixes this issue?

MJ666, 2017-04-10:
Here are the two settings I have used when seeing the problem:

`# Betaflight / PIKOBLX 3.1.7 Apr  3 2017 / 22:16:26 (e1c4b5c)
name AcroWhoop

feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature MOTOR_STOP
feature TELEMETRY
serial 1 4 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
aux 0 0 1 1700 2100
aux 1 28 2 1700 2100
aux 2 4 0 1300 1700
aux 3 1 0 1700 2100
set min_throttle = 1000
set use_unsynced_pwm = ON
set motor_pwm_protocol = BRUSHED
set motor_pwm_rate = 32000
set serialrx_provider = SBUS
set sbus_inversion = OFF
set acc_trim_pitch = -4
profile 0

set p_pitch = 80
set i_pitch = 37
set p_roll = 65
set i_roll = 37
set d_roll = 35
set p_yaw = 103
rateprofile 0
rateprofile 0

set rc_rate = 114
set rc_rate_yaw = 101
set roll_srate = 69
set pitch_srate = 69
set yaw_srate = 53

# `

`# diff
# Betaflight / PIKOBLX 3.1.7 Apr  3 2017 / 22:16:26 (e1c4b5c)
name PikoBLX110 DSHOT
resource MOTOR 1 A07
resource PPM 1 NONE

feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature MOTOR_STOP
map TAER1234
serial 2 64 115200 57600 0 115200
aux 0 0 1 1700 2100
aux 1 27 2 1700 2100
aux 2 1 0 1300 1700
aux 3 2 0 1700 2100
aux 4 12 6 900 1700
rxfail 3 h
set min_check = 1060
set max_aux_channels = 8
set min_throttle = 1060
set motor_pwm_protocol = DSHOT600
set serialrx_provider = SPEK2048
set vbat_warning_cell_voltage = 34
set pid_process_denom = 2
profile 0

set p_pitch = 45
set i_pitch = 40
set d_pitch = 18
set p_roll = 45
set d_roll = 18
set p_yaw = 72
set i_yaw = 40
rateprofile 0
rateprofile 0

set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# `

chime13, 2017-04-10:
@Arakon - anti gravity did nothing to alleviate the problem. I had to go back to 3.1.6 in order to fly correctly again. My anti-gravity setting is just something I'm playing around with, and 4 seems to work for me.

borisbstyle, 2017-04-10:
Nobody can run a blackbox of it? The only common thing is that both are small micro quads right?

Eganwp, 2017-04-10:
I had this exact same issue happen on one of my quads last week with 3.1.7. Twice it happened on back to back 5s batteries around 2/3rds throttle punching to full. I can't get it to happen on 4s though, so I wrote it off as just a classic desyncs not knowing what may have caused it. Never had it happen before on that quad and it has over 50 flights on it on 3.1.6 previously to that. Unfortunately I didn't have my BB logging while it happened. I wouldn't have thought twice about it until reading and seeing others having the same issue. I have been flying with Anti_gravity Gain of 4.0 on all my quads on 3.1.6 and 3.1.7 and the desync happened with it enabled so I don't think that's the fix. Who knows, I may just have a bad ESC, I'll have to do more testing.

chime13, 2017-04-10:
Hi @borisbstyle , Sorry no BBox here. 
Just to confirm:
Prior to the upgrade - all of my builds were tuned and were predictable in behavior.
The issue happened to all 3 of my FCs after upgrading to 3.1.7. (2- F1type and 1- F3type)
All of these builds are at or under 100mm.
I'm running 8.5mm brushed motors on 55mm props (Afunta and Ladybird)
All of my builds are under 36grams without battery.
The issue went away the moment I restored 3.1.6 on all my builds.
HTH

borisbstyle, 2017-04-10:
@Eganwp Well if it was only happening on 5s for you it may simply just be desync.
This seems to happen now only on these small builds

@chime13 do you fly with airmode enabled?

chime13, 2017-04-10:
@borisbstyle - no AIRMODE (yet) and I also disable the ACC to save battery juice. Gyro loop settings are 2khz and PIDs at 1khz.

HackIT, 2017-04-10:
I've met some funny ppl in the multirotors community, that was hilarous...
In order to demystificate all the wrong IDEAS many ppl transmitting...
AIRMODE is enabled AUTOMATICALLY when no other mode is selected it usually happens when your throttle stick reaches a certain input (1350~) and by this it's now enabled permanently for the flight!

Regards.

Arakon, 2017-04-10:
Actually, it's disabled by default in 3.1.7. You have to activate it as feature or by switch, otherwise it's off permanently.

HackIT, 2017-04-10:
@Arakon Meh, so many changes... why doesn't it now?

MJ666, 2017-04-10:
For me all three builds are tiny. The problem happens with and without Air Mode enabled always at fast forward speed.

thexboxblasater, 2017-04-10:
One of @MJ666 builds is a brushless if I see correctly though.
I will try the 3.1.7 on my micro brushed which is now running 3.1.6, to see if the problem occurs.

Rossbow, 2017-04-17:
I am having the same issue.
Micro Lux F4 (BluejayF4 Target), Lumenier 10a 4in1, DYS1104-7500kv, 2s Lipo
Applied full throttle in level mode when it happened

chime13, 2017-04-18:
@Rossbow - All I can suggest is that you backup your settings, downgrade to 3.1.6 or 3.1.5 with a full chip erase, then restore. If you're familiar with CLI, I would suggest that you type
 'Dump all showdefaults'  
and compare your values with the defaults. I had a case in which there was some crazy data stored that I know I didn't set and don't know if it could cause errors or the like.

borisbstyle, 2017-04-18:
It kind of looks like desync of one esc where pids have to work hard to correct it.
There is nothing weird to see on the log.

19Gerhard85, 2017-04-18:
[DVR Footer](https://youtu.be/PXa_AQBqyb0)

Rossbow, 2017-04-18:
@chime13 -  Thanks.  Yes I am fimilar with CLI and all facets of BF, I want to contiinue to trouble shoot this version so I won't downgrade for now

Rossbow, 2017-04-18:
OK so today I tried removing the filters one by one and it was all fine.  So then I put a few batteries through and did full throttle punches and did not have an issue, Now I have only run 2 batteries so far but it is promising

set d_lowpass_type = PT1
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set d_notch_hz = 0

borisbstyle, 2017-04-18:
@nathantsoi Please enable Airmode. IT was not enabled in your log

Dodgetweaker, 2017-04-19:
I figured out tonight that this has been the issue with my tiny whoop with a beebrain.  The whoop would lose stability only under full throttle in angle, horizon, and acro mode.  I reverted back to 3.1.6 and all is well again.

borisbstyle, 2017-04-19:
I think you also didnt have airmode enabled I would guess

19Gerhard85, 2017-04-19:
@borisbstyle So Airmode is mandatory in BF 3.1.7?

Dodgetweaker, 2017-04-19:
@borisbstyle I tried turning on airmode and I don't recall it helping.  I tried just about everything I knew of over the past 3 days.  I even replaced every single hardware component one at a time before I noticed that the FC that worked was running 3.1.6 (I thought they were the same).  I am willing to confirm the issue again if it helps.  Unfortunately, I can't record any data with the beebrain.  Please let me know what I can do to help.  I plan to stay on BF3.1.6 for now.

MJ666, 2017-04-19:
Air Mode did not made a difference for me.

borisbstyle, 2017-04-19:
@Dodgetweaker
I actually have a beebrain target so I can test.

All these issues only seem to occur with brushed motors it seems.
The one from @nathantsoi above is not brushed one and it clearly shows drops in Iterm on 0 throttle.

MJ666, 2017-04-19:
Also one of my build is not brushed but only 110mm. It's now flying with 3.1.0 without issues. In the link below are the details for this build. The configuration used for BF 3.1.7 was the same.

http://www.multiwii.com/forum/viewtopic.php?f=12&t=4893&start=100#p68854

HackIT, 2017-04-19:
Gyro spikes makes unfortunately a motor or two to stop or almost stop... which makes twitches from HELL!
You have to call an exorcist to exorcize your FCs... May Jesus help you.

MJ666, 2017-04-19:
@HackIT why all my tree copters fly well with a former firmware and the same settings? This is also the case for a few more users which are downgraded after having problems with 3.1.7?

HackIT, 2017-04-19:
@MJ666 I did too, speaking of downgrading... it was with an older version 3.1.5 to 3.0.1...
Nothing helped... same twitches from hell... till I started reading the gyro graph and discovered that the IMU6500 doesn't like to be shaked by motors continuously... for my case the gyro twitches on Z axis the gyro output shows that the quad rolled(Z axis) enough to be upside down but the quad didn't move... since it has no props and it is leveled on the table...

MJ666, 2017-04-19:
Many of my boards have MPU6500 and work without problems. Most of them are in brushed  builds mounted with double sided sticky pads but also have some which are realy hard mounted. The tree copters with the problem all have a MPU6000 and this is known to be less sensitive for vibrations.

HackIT, 2017-04-19:
@MJ666 it depends heavily how you flying your quad... I didn't had twitches while hovering or flying forward quietly... it happened only when I attack the air (throttling, rolling and pitching sharply)

nathantsoi, 2017-04-23:
thx for the advice, @borisbstyle -- airmode fixed it on both the x120 and x200, exact same setup, this time w/ airmode (looks like the blackbox log viewer is reporting my air mode switch as super expo.... but that's really airmode)

new logs + video: https://www.dropbox.com/sh/knt8lkq9habrjcx/AACqps54l5LfHdjfvZsMw3kGa?dl=0

rnelias, 2017-05-03:
following since I'm having a similar problem. Will edit a video later today and post.

HackIT, 2017-05-05:
I've exorcized all my FC's (Jesus didn't hear my scream) by enabling airmode permanently putting airmode full range on a AUX from mode tab...
I have to say that the guy who put the airmode feature in the configuration tab didn't do the job correctly...
The twitches from hell came from there... thank you or not... from this I decided to fork for my personal use.
Thank you for the source code.

rnelias, 2017-05-05:
I think I've found out what was happening with my quads...

With the change to 3.1.7 and to Omnibus, I've filled out the fields in the configuration tab one-by-one by hand (I usually copy diffs from version to version). In the "MISC" tab, there are just two fields: "Craft's Name" and "FPV camera angle", as I was setting up my OSD, I've found both should be filled out and, as they're "miscellaneous" settings, they should not be so important... well... I don't think so. Today I finally understood that these naive field is not so naive and it was compensating my natural yaw/roll inputs putting more yaw on my commands and making the craft fly weirdly. I've already set the fpv camera angle back to zero and will make some tests tomorrow. For now, PLEASE, remove this parameter from the MISC group or put an observation close to it.

fotoingo, 2017-05-09:
I did have the same problem with my Blade Torrent 110 !
Updated from BF 3.1.5 to 3.1.7 and boom. When doing punch outs the quad flipped diagonial into the ground. (Like if the left right motor didnt have enough power)
Tried everything from switching the motors, calibrating the escs and reflashing BF 3.1.7 (full eraese) or trying different settings....

Saw this post here and flashed to 3.1.6 -> Problem is gone !!!

rnelias, 2017-05-09:
Take a look in the following BB log that illustrate what I'm having here with 2 quads with almost identical setups (described bellow):

Both have:
Omnibus F4 v2 BF 3.1.7
Kiss ESC 24a RC9 (dshot600 8k/8k)

Quad 1 (Chameleon) uses BrotherHobby 2206/2300kv

Quad 2 (QAV-210) uses Hyperlite 2206/2450kv

The log bellow is from quad 2 where the problem is even worse but happen with both. I could assume it's an esc or motor with problem but, with two (almost new) quads is a little bit weird...

[image]

chime13, 2017-05-10:
@melias - did you try to revert back to a known-good version "3.1.X" to see if the problem goes away? As I originally reported, I returned to 3.1.6 and the problem wit away. Others have reported similar success reverting back to 3.1.6 or 3.1.5.

rnelias, 2017-05-10:
@chime13 not yet but will do it and repeat the test tomorrow

@ledvinap I'm sending attached. user "etracer" gave me an interpretation on RCGroups but I refuse to accept as the only explanation since these two quads have weird behavior and have almost the same hardware. I also have other 2 quads flying SPRF3 with littlebee (multishot) and BF3.1.7 with no issues at all. 

Here's the post where etracer interpreted my BB log. If you look close to frames #651732 session 3/6 you'll see motor #2 stuck at 100% power, supposedly reacting from a little bit of roll input. 
[BLACKBOX_LOG_QAV210_20170509_175020.zip](https://github.com/betaflight/betaflight/files/989878/BLACKBOX_LOG_QAV210_20170509_175020.zip)

https://www.rcgroups.com/forums/showpost.php?p=37488087&postcount=48226

ledvinap, 2017-05-11:
@rnelias : It seems that etracer explanation is right.

Try enabling setpoint graph, set graph expo to 100% and compare setpoint with gyro readout. First part of maneuver is saturating motor3 to get enough roll+pitch, second part is saturating mortor2 to slow down rotation (in both axes).

Strange thing is that pitch rate is (almost) not decreasing with motor2 running at full power. I would expect the slope to be comparable with part where pitch rate increases ...
Maybe there is huge nonlinearity in force generated by motor and motors at ~40% are generating almost same force ast motor at 100%.

Blackbox viewer IMU algorithm seems to fail badly in this case, estimated pose is far from gyro readings. Porting new quaternion based IMU to viewer will probably help a lot. 

Can you upload video too ? I have hard time visualizing actual quad rotation during the maneuver

HackIT, 2017-05-11:
@rnelias, I had same problem... it was happening very often when I had used perma airmode feature from config tab which is totally wrong...(you must set the mode yourself into mode tab...)
Well so, I've done some test using real time gyro/accel reading from the interface...
Then discovered that the gyro were so disturbed by the noises that sometimes the graph goes crazy a like the quad had rolled a side...
The problem is that the frame reacts like a guitar cord... it resonates through the entire frame...
What fixed this is softmounting the arms(frame using separate arms...) with some red double sided tape...
My fc was already softmounted between 2 M2 rubber rings and nylon standoff.
My motors are not softmounted.

To illustrate my setup:
Realacc X5r 220mm
Cicada 18A (Blheli_s startup power 1.25)
Cobra 2205 2300kv
Lumenier Luxx Like v1 (8k/8k dterm filter off)
I also use 2 25V 470uf capacitors on my PDB to smooth voltage peaks...

rnelias, 2017-05-11:
@ledvinap I will save a lower resolution version in my dropbox and share the link. Sadly, in my last flight I've pushed a lot in the throttle and the problem happened a lot but the Omnibus memory was full and could not save the data :( . In the flight that I had shared the log I could only notice the problem on that range that I mentioned. I can only assure it has something to do with hard throttle punch outs. It's even worse with my quad using a bigger kv (2450)

@HackIT both quads have rubber bobbins holding the flight controller but the motors are hardly mounted. The Chameleon (2206/2300 Returner motors) is more rigid than the QAV210 (4mm plates / 3mm).

Well... I wouldn't like to put it in this way but I don't consider myself a noobie in flying mini quads. Of course, I can't be considered a "pro" pilot but I'm pretty capable to feel when something goes wrong. I don't know if this is a hardware issue, a software issue or, even worse, a configuration issue (I don't follow all the develepments behind BF so close... you guys are amazing in developing new things). What I know is that the same problem happen with two quads with almost the same setup (only motors and frames are different) flying BF3.1.7. I also have a QAV-X with LittleBee 20a and SPRF3, also BF3.1.7 but Multishot 4k, and it flies very well with no issues at all.

Today is raining here but I'll step back in BF version and see if the problem persists...

HackIT, 2017-05-11:
@rnelias, believe me, none gyros can hold accuracy when it resonates as a guitar cord!...
Even by using softmount on your FC it still getting a lesser amount of vibration... that's why you have to kill the resonance of your frame... (only if it is resonating...) use something a like a screw driver and hit the arms and with the other hand you hold the quad by the top plate if you feeling the resonance(not the hit, "RESONANCE" it's lil vribrations persisting after an hit!) after hitting it you have to softmount something to kill this resonance... also check your ball bearing(if they aren't smooth it's bad...must change them) and props(if they aren't balanced it make noises too)!

Hope it helps.

rnelias, 2017-05-11:
@HackIT Chameleon and QAV210 are unibody and softmount is only possible in the motors not in the arms. Bearings are not the case in this issue since both quads have less than 30 flights and the motors are smooth as silk.

HackIT, 2017-05-11:
@rnelias, Well was just trying to help.
Hope you find the way to fix this issue.
I gave you all the tips that made my kwads flies like birds without twitching.

rnelias, 2017-05-11:
@ledvinap here's the vid corresponding to the BB log I've sent previously

145mb https://www.dropbox.com/s/y1xxgh4xahbmk66/bb-2450kv.mp4?dl=0

and here's the last flight I've mentioned. In this flight, I've tried to push the throttle more than I usually do and the quad was completely unpredictable, however, the log was not recorded since the memory was full. But I think it's noticeable that the mini was not doing what I was trying to do...

65mb (I've crashed in the end) https://www.dropbox.com/s/rebcgtyy1ofmi85/2450-fight-to-fly.mp4?dl=0

just in case, here's my YT channel https://www.youtube.com/rnelias

heikkoe, 2017-07-13:
I too experienced something like this - upgraded to 3.1.7 and started getting the random glitches. Log attached, session 3/3 in the end has a few of those where motor1 just flats out to max. 
[BLACKBOX_LOG_20170711_105853.bfl.zip](https://github.com/betaflight/betaflight/files/1144222/BLACKBOX_LOG_20170711_105853.bfl.zip)
Downgraded to 3.1.6 and can no longer reproduce.

dronebrulee, 2017-07-15:
Same here - Betaflight / SPRACINGF3EVO 3.1.7, brushed micro build would pull hard to the right on throttle punch.  Downgraded to 3.1.6 and all is works well now.  My CLI diff:

# Betaflight / SPRACINGF3EVO 3.1.7 Apr  3 2017 / 22:28:08 (e1c4b5c)
name KK 100 brushed

feature -RX_PPM
feature -TELEMETRY
feature -CURRENT_METER
feature -RSSI_ADC
feature -BLACKBOX
feature -TRANSPONDER
feature -SDCARD
feature RX_SERIAL
serial 1 64 115200 57600 0 115200
aux 0 0 0 1500 2100
aux 1 1 1 1700 2100
aux 2 13 2 900 1500
rxrange 0 1188 1812
rxrange 1 1252 1826
rxrange 2 1032 1966
rxrange 3 1110 1930
set min_throttle = 1034
set serialrx_provider = SBUS
set deadband = 3
profile 0

rateprofile 0
rateprofile 0

wuub, 2017-09-27:
Since a lot of people are pitching in with their 'me too' info. I want to add a success story instead 

 I too experienced the flip-on-throttle in both angle and acro mode on a freshly upgraded (3.1.7) babyhawk. I could have sworn I had AIRMODE enabled. When I found this issue I was disappointed, that it does not help me. 

Then I went and double checked, lo and behold, due to a small bug in my config, AIRMODE was **not being enabled** . 

I permanently (feature) enabled AIRMODE and ANTI-GRAVITY.
Flew 3 packs in the morning, no flips, no drift, no wobble during punchouts, both angle and acro. Straight up, at different angles, different speeds, fresh battery, depleted battery just before landing.  No flips in sight.

Then I remembered my **brushed** SP3EVO quads expierienced similiar flips in ANGLE. I dismissed it then as a strange accelerometer mishbehaviour and flew ACRO.  Now I'm 99% sure they were configured to enable airmode only for acro, and as such this problem manifested itself in angle.

So thank you @borisbstyle. It's my config that was broken, not hardware :)

closed 2018-04-21 by DieHertz