betaflight/betaflight #2854: Problem with 3.1.7 on three PikoBLX targed based builds

MJ666, 2017-04-10:
Build three new quads last week. Two are using an AcroWhoop and one is using a PikoBLX FC. All tree builds expose some uncommanded tendency to do flips. The two whoop style copters always crashed but the 110mm brushless build could be recovered many times since it was an short hick up only. This always happend with higher flying speeds never during normal hover or slow flying. Initially it feeld like an motor desync but this does not made sens fro the brushed copters. Also the brushless motors of the 110mm build running smooth with the configurator motor tab on the bench via the full range. After downgrading to 3.1.0 with the same settings all copters are flying fine. Have not done testing with the versions in between. Has someone experienced similar problems?

MJ666, 2017-04-10:
Sounds quite similar to #2852

closed 2017-04-10 by mikeller

borisbstyle, 2017-04-10:
What is different on PikoBLX FC to cause this?

MJ666, 2017-04-10:
Likely not specific for the PikoBLX only taking #2852 into account. I only had 3 different copters and all are running with the PikoBLX target.