betaflight/betaflight #2860: Throttle jump and lock on different values

Otakee, 2017-04-11:
Hi everyone,

I used a new setup with : 
- Flip 32 F4 : Airbot 3.1.6, 3.17 and another version
- Little bee ESC Spring 35A : 16.6 Bheli
- F40 Pro motor 2600kv

My throttle jump and lock after 1500 value and jump directly on random value between 1500 and 2000 during 1 or 2 seconds. I tried change max_throttle, min_throttle, min_check, max_check, witch multishot, oneshot and dshot, acro, horizon mode but behavior also persit.

For example : 

[image]

[image]

Do you have any idea ? it's bug with new FC F4 ?

borisbstyle, 2017-04-11:
you need to post the entire log file and not just screenshot

Otakee, 2017-04-11:
Sorry, here this file from blackbox.

[BLACKBOX_LOG_20170409_172400.txt](https://github.com/betaflight/betaflight/files/913111/BLACKBOX_LOG_20170409_172400.txt)

borisbstyle, 2017-04-11:
looks like failsafe. 

How is your failsafe configured?

Otakee, 2017-04-11:
@borisbstyle Yes, failsafe it's default settings configured on tab

Arakon, 2017-04-11:
Definitely failsafe.. you lost connection to your transmitter. Not a bug, simply bad signal/antennas/receiver.
[image]

Otakee, 2017-04-11:
@Arakon, ok ! thanks you so much for the investigation.

The behavior of throttle is due to failsafe ?

Arakon, 2017-04-11:
Yes, it is. Your FC keeps getting the same signal while it cut out, and then suddenly a new one when the connection is back.

Otakee, 2017-04-11:
@Arakon, it's very strange with 3S lipo i haven't any problem with my receiver.

Otakee, 2017-04-13:
I test with another radio / receiver and problem persists. I think a FC create a voltage drop during high throttle and the receiver drop signal.

I'll add a bec between power FC and i will let you know soon.

Otakee, 2017-04-20:
With another RX problem is solved !

closed 2017-04-20 by Otakee