betaflight/betaflight #2861: Proposed feature: PID tuning on-the-fly

RipperDrone, 2017-04-11:
Is it even possible / can be thought of?

Advantage: Not having to land/relaunch multiple times during PID tuning.

Best solution would be to have this feature implemented in a way that does not require a complex sequence of buttons/switches to change & save values, so pilots can do it with ease while keeping their goggles on.

Maybe I'm just dreaming here, but it does not sound too much out of this world for me to implement... just have no idea if writing values back to FC EEprom will not CPU / can't be done in the background whilst running the PID loop at the same time etc. Maybe just needs a different temporary RAM buffering system?

Thoughts?

Arakon, 2017-04-11:
That's already possible by using switches or potentiometers set in the "Adjustments" tab of the configurator.. i.e. 3 Position switch to increase and decrease P, I or D.

martinbudden, 2017-04-11:
Yes, this is already possible. Joshua Bardwell has done a video: https://www.youtube.com/watch?v=JpnCMLmxme0&t=127s

RipperDrone, 2017-04-11:
Great, never thought about it since I'm enjoying the PID-over-Taranis feature for tuning. Almost forgot about the existing potentiometer option. Thanks both for reminding! Will close this issue now 🙂

closed 2017-04-11 by RipperDrone