betaflight/betaflight #2864: Cannot disable angle mode

Camoleopard, 2017-04-12:
I just flashed my SPFRacing F3 board in my wizard x220 to Betaflight 3.1.7 today. After flashing it, I went through and put my previous settings in. I went to test to see if they were working and found that airmode was no longer working. No matter what I did, permanently enabling airmode etc. I could only use angle mode, or some mode that used the gyro, because moving the quad around resulted in different speeds on the motors. I then reverted back to Betaflight 3.0.1, and the issue persisted. I don't believe it is my board because it was working before the update. Any help?

AndersHoglund, 2017-04-12:
"...put my previous settings in." Manual or pasting from an old diff or dump? 
Never restore settings across different versions using a complete dump. Restoring from diff may work better, but things change between versions and this may not work either. In 3.1.7 the mode identities has changed and that may break any old mode settings, restore manually from the GUI.
I your case, I would advice you to Reset Settings and start over. Configure everything from scratch.

Camoleopard, 2017-04-12:
Hey, 
I should have been more clear, I put in my old settings manually. As a side note, I got the blackbox data and it says I am in angle mode even though I have enabled permanent airmode. Also, when I enter angle mode, my Props are constantly spinning, is this intended? Because in previous versions in angle mode the props would not be spinning the whole time.

AndersHoglund, 2017-04-12:
You can have AirMode and Angle at the same time. Two different things, a common confusion though. Props spinning when armed is what you want, regardless of Angle or Horizon. A steady idle and a bit more to avoid stalling and/or de-syncing motors when doing zero throttle manoeuvres. I.e. for AirMode to have something to work with and to be able to stabilize the rig at zero throttle.

closed 2017-05-24 by mikeller