betaflight/betaflight #2865: DSHOT idle percent behavior

19Gerhard85, 2017-04-12:
Hello guys,

yesterday I found a strange behavior of the idle percent.

I've determined the lowest idle speed of my motors following the instructions of Cheredanine (from the Wiki page https://github.com/betaflight/betaflight/wiki/DSHOT-ESC-Protocol).
I came to the same value of 1040 which are 4%.

But with this value, the FC won't arm / the motors won't spin. I've changed it back to the default of 4.5 and everything worked fine.

My motors (EMAX RS2205S 2600KV on TBS 25A) start spinning at around 1010).

borisbstyle, 2017-04-12:
Digital idle offset and normal pwm values are not necessarily the same.
Pwm can be calibrated and dshot not

19Gerhard85, 2017-04-12:
Ok, thanks!

closed 2017-04-12 by 19Gerhard85