betaflight/betaflight #2868: anyfc : no device detected on i2c bus

urlu75, 2017-04-12:
Got the opportunity to try a [F7 board](http://www.banggood.com/STM32F745-100lqfp-216MHz-MPU6000-SPI-F7-Flight-Controller-for-FPV-Racing-Support-Betaflight-p-1137386.html).
No i2c device detected (MS5611 or HMC5883L). 
with 3.1.7 : 2 errors are reported on the GUI at boot time ith or without i2c sensor connected.
with 3.1.6 : 1 error reported

sambas, 2017-04-12:
ms5611 baro is connected to internal I2C (unpopulated on this BG board), mag (and external i2c bus) is not enabled in betaflight. INavFlight has support for external devices

urlu75, 2017-04-12:
thanks for your reply. Closing the issue.

closed 2017-04-12 by urlu75

sambas, 2017-04-12:
Quick question, did the board come empty or some firmware in it?

urlu75, 2017-04-12:
BF 3.1.6 on it.
And about INavFlight, it is not starting. I will double check tonight and open an issue if needed.

sambas, 2017-04-12:
Check that you have ANYFCF7 firmware on inav, inav also has anyfc which is earlier F4 board

urlu75, 2017-04-12:
yes thanks, better after compiling ANYFC7. but no i2c sensor detected