betaflight/betaflight #2878: [OmnibusF4] BlackBox on SDCard

celiandu61, 2017-04-13:
Hey, i tried to upgrade my omnibus f4 yesterday, i used `dump` to find the firmware ( OMNIBUSF4 in my case ) and i launched the flash. Everything is working perfectly execepted the blackbox. I can't choose to use my SDCard to save my logs ( but i can before upgrade ). My upgrade is from 3.0.1 to 3.1.7. The feature SDCard is enabled. 

I see another firmware named OMNIBUSF4SD but before upgrade in the dump it was writted OMNIBUSF4. In the wiki my omnibus f4 pro v2 does'nt have any firmware in the table.

> # version
> # Betaflight / OMNIBUSF4 3.1.7 Apr  3 2017 / 22:14:59 (e1c4b5c)
> 
> # name
> name -
> 
> # resources
> resource BEEPER 1 B04
> resource MOTOR 1 B00
> resource MOTOR 2 B01
> resource MOTOR 3 A03
> resource MOTOR 4 A02
> resource MOTOR 5 A01
> resource MOTOR 6 A08
> resource MOTOR 7 NONE
> resource MOTOR 8 NONE
> resource MOTOR 9 NONE
> resource MOTOR 10 NONE
> resource MOTOR 11 NONE
> resource MOTOR 12 NONE
> resource SERVO 1 NONE
> resource SERVO 2 NONE
> resource SERVO 3 NONE
> resource SERVO 4 NONE
> resource SERVO 5 NONE
> resource SERVO 6 NONE
> resource SERVO 7 NONE
> resource SERVO 8 NONE
> resource PPM 1 B14
> resource PWM 1 B14
> resource PWM 2 B15
> resource PWM 3 C06
> resource PWM 4 C07
> resource PWM 5 C08
> resource PWM 6 C09
> resource PWM 7 NONE
> resource PWM 8 NONE
> resource LED_STRIP 1 A01
> resource SERIAL_TX 1 A09
> resource SERIAL_TX 2 NONE
> resource SERIAL_TX 3 B10
> resource SERIAL_TX 4 NONE
> resource SERIAL_TX 5 NONE
> resource SERIAL_TX 6 C06
> resource SERIAL_TX 7 NONE
> resource SERIAL_TX 8 NONE
> resource SERIAL_TX 9 NONE
> resource SERIAL_TX 10 NONE
> resource SERIAL_TX 11 NONE
> resource SERIAL_TX 12 NONE
> resource SERIAL_RX 1 A10
> resource SERIAL_RX 2 NONE
> resource SERIAL_RX 3 B11
> resource SERIAL_RX 4 NONE
> resource SERIAL_RX 5 NONE
> resource SERIAL_RX 6 C07
> resource SERIAL_RX 7 NONE
> resource SERIAL_RX 8 NONE
> resource SERIAL_RX 9 NONE
> resource SERIAL_RX 10 NONE
> resource SERIAL_RX 11 NONE
> resource SERIAL_RX 12 NONE
> 
> # mixer
> mixer QUADX
> 
[... 468 lines cut ...]
> set i_level = 50
> set d_level = 100
> set p_vel = 55
> set i_vel = 55
> set d_vel = 75
> set level_sensitivity = 55
> set level_limit = 55
> rateprofile 0
> 
> # rateprofile
> rateprofile 0
> 
> set rc_rate = 121
> set rc_rate_yaw = 100
> set rc_expo = 0
> set rc_yaw_expo = 0
> set thr_mid = 50
> set thr_expo = 18
> set roll_srate = 70
> set pitch_srate = 70
> set yaw_srate = 70
> set tpa_rate = 10
> set tpa_breakpoint = 1650

Thx for your help, sorry for my bad english :) I you have any question, ask them.

nathantsoi, 2017-04-13:
If your omnibusf4 has an sdcard, use the omnibusf4sd target. Don't know why
it used the other target before. Maybe a custom firmware?

celiandu61, 2017-04-13:
Idk, i bought my card here : https://www.fpv4drone.com/en/flight-controllers/322-carte-de-vol-airbot-f4.html?search_query=omnibus+f4&results=1

I can contact the seller tomorrow to try to get information about that. I can't have hardware issue if i flash a bad driver ?

nathantsoi, 2017-04-13:
You're probably fine flashing either omnibusf4 or the omnibusf4sd firmwares
to test, worse thing that can happen is you need to restore the firmware
with an st-link.

But the Omnibusf4sd target on GitHub or in the configurator is definitely
the correct one for any Omnibus f4 with an sdcard. For folks with an
omnibusf4 that doesn't have an sdcard, the omnibusf4 target is the one to
choose.

celiandu61, 2017-04-13:
I tried 5 min ago but it stay at flashing stage. But zadig said WCID are wrong, i will retry tomorrow with my computer to be sure. I hope it's not my card :/

yosemitez, 2017-04-13:
I had this problem on a rg ssd f3, try
Set feature BLACKBOX

celiandu61, 2017-04-14:
Ok, i flashed the OMNIBUSF4SD. But now the serial port is'nt working :/ I can't open the serial port and the dfu is detected as unknow device now :'( 

> No configuration received within 10 seconds, communication failed

I have to flash it with st-link ? If yes can you explain to me how to do ? I have to bought something or it's possible with an arduino ?

nathantsoi, 2017-04-15:
Oh yeah, you don't even need an STLINK, just hold the boot button to get it
into DFU mode, even if the wrong firmware is flashed, that will still work.

If the drivers are broken try:
https://impulserc.blob.core.windows.net/utilities/ImpulseRC_Driver_Fixer.exe

And try flashing the other hex.

Could you also email me a photo of your board? Where did you get it?

celiandu61, 2017-04-15:
Ok, i tried on 2 computers yesterday and it was detected as unknow device in dfu ( it worked before flashing omnibusf4sd ) and this morning it working first time on my computer xD 

I'm now in OMNIBUSF4 but the yosemitez's method doesn't work ...

I bought it here, you have a photo in the website ( exactly the same as mine ).
https://www.fpv4drone.com/en/flight-controllers/322-carte-de-vol-airbot-f4.html?search_query=omnibus+f4&results=1

celiandu61, 2017-04-17:
I just retried with the OMNIBUSF4SD and this time it's work ! Thx for your help @nathantsoi

closed 2017-04-17 by celiandu61