betaflight/betaflight #2879: DSHOT protocol implementation cannot reach 100% throttle

lazd, 2017-04-13:
I was looking at the DSHOT ESC signals with my oscilloscope, and I noticed the least significant bit of the 11 bits that indicate throttle value was never turned on, no matter what throttle value I selected in the motors tab. I realize that's because Betaflight's internal representation of throttle only has 1000 steps, so we're only using half of the throttle resolution the DSHOT protocol is capable of and therefore only getting even numbers, but it's resulting in an off-by-one error...

## Expected Behavior

Betaflight sends 2047 for 100% throttle.

## Actual

Betaflight sends 2046 for 100% throttle.

## Steps to reproduce

In this screenshot, the motor output value was set to 1001 in the Motors tab. This results in a decoded value of 48, which is the lowest non-zero throttle value in DSHOT (as 0 means disarmed and 1-47 are reserved for commands). If you consider 0 the 1st throttle step, this is the 2nd of 2000 total throttle steps, as expected.
[image]

Of course, increasing the motor output value to 1002 results in a decoded value of 50, as expected, since we're only using half the resolution of the protocol.

However, in this screenshot, the motor output value was set to 2000 (100% throttle) in the Motors tab, but only the first 10 bits are on, resulting in a value of 2046 -- the 1999th of 2000 throttle steps -- when the actual maximum throttle value should be 2047:
[image]

As a result, Betaflight is never sending 100% throttle to ESCs when using the DSHOT protocol. Additionally, a motor output value of 1500 does not result in exactly 50% throttle -- both are off by 1.

### Why?

Inside of `convertExternalToMotor()`, we have:

```c
motorValue = externalValue <= EXTERNAL_CONVERSION_MIN_VALUE ? DSHOT_DISARM_COMMAND : constrain((externalValue - EXTERNAL_DSHOT_CONVERSION_OFFSET) * EXTERNAL_DSHOT_CONVERSION_FACTOR, DSHOT_MIN_THROTTLE, DSHOT_MAX_THROTTLE);
```

`DSHOT_MAX_THROTTLE` is defined as 2047, but because `externalValue` is multiplied by `EXTERNAL_DSHOT_CONVERSION_FACTOR` which has a value of `2`, for 2D and positive direction 3D flight, the motor output value can never be odd and can therefore never be equal to 2047.

Currently, we get the following values for 2D flight:

2000 => 2046
1501 => 1048
1500 => 1046
1499 => 1044
1001 => 48
1000 => 0

And we get the following values for 3D flight:

2000 => 2046
1501 => 1048
1500 => 0
1499 => 51
1001 => 1047
1000 => 0

As you can see, maximum throttle is never achieved for 2D or 3D flight in the positive direction, however, maximum throttle is achieved for 3D flight in the negative direction (1047 is maximum throttle in the negative direction). Additionally, the middle throttle value of 1500 results in a signal of 1046, which is actually off by 1 from the actual middle throttle value in DSHOT (1047).

### Proposed solution

If we simply add 1 to `motorValue` for normal 2D and 3D flight alike, we get the following values:

For 2D flight:

2000 => 2047
1501 => 1049
1500 => 1047
1499 => 1045
1001 => 49
1000 => 0

For 3D flight:

2000 => 2047
1501 => 1049
1500 => 0
1499 => 51
1001 => 1047
1000 => 0

After this change, all throttle values sent to the ESC will be odd, we'll have the maximum throttle values possible for both 2D and 3D flight (equally in both directions), and the middle ESC throttle value of 1500 will result in the expected DSHOT value of 1047. 

However, we will no longer be able to send the minimum throttle value for 2D flight. That said, because `digital_idle_percent` is never a non-zero number (motors would not spin), it's very unlikely minimum ESC throttle value of 48 will be sent to the ESC, but it is very common that the maximum ESC throttle value of 2047 will be sent. As such, I believe we lose nothing by making this change.

borisbstyle, 2017-04-13:
I think you are testing through configurator and that does the conversion.
In real flight that conversion is not happening

DieHertz, 2017-04-13:
Yep, I've noticed it in Configurator too.
But in real flight `float` variables are used and computation resolution is limited only by IEEE-754 specs.

lazd, 2017-04-13:
Yes I'm testing in the configurator. I'll see what happens when arming and throttling up and report back.

lazd, 2017-04-13:
@borisbstyle @DieHertz I just tested 3.1.7, and when arming and throttling up, the problem is not present -- I see full throttle represented correctly on my scope. That said, is this still an issue that should be addressed?

borisbstyle, 2017-04-13:
@lazd It is a little limitation in how the configurator communication is done. Configurator also has 1000 point of resolution now and dshot has twice as much. You also cant do all steps through it.

lazd, 2017-04-13:
@borisbstyle understood. The PR is there if it's worth merging. Thanks!

commit 06dba20e (2017-04-17): Fix DSHOT off-by-one error preventing maximum throttle, closes #2879
closed 2017-04-17 by borisbstyle