betaflight/betaflight #2883: 3.1.7 - SP Racing F3 - no yaw response

MimaRaza, 2017-04-14:
After flash to 3.1.7 i can't stear yaw.

Stearing throttle, pitch and roll works.

Connected on PC it works.

Only setting i changed in cli is yaw motor direction

hydra, 2017-04-15:
iirc motor direction in 3.1.7 hould be -1 or +1, a setting of 0 will result in no yaw.

MimaRaza, 2017-04-15:
So standard setting is:

yaw motor direction - 1 

and if i wont to change it i change it like that:

set yaw motor direction + 1

MimaRaza, 2017-04-15:
This is what i tryid know, can't get back to standard setting

# set yaw_motor_direction + 1
Invalid name

# set yaw_motor_direction - 1
Invalid name

# set yaw_motor_direction - + 1
Invalid name

# set yaw_motor_direction  = + 1
yaw_motor_direction set to 0
# set yaw_motor_direction = + 1
yaw_motor_direction set to 0
# set yaw_motor_direction = - 1
yaw_motor_direction set to 0
#

Arakon, 2017-04-15:
Remove the space. I changed the yaw direction on 3.1.7 just yesterday with no issue.
It has to be -1 and +1, not - 1 and + 1.

MimaRaza, 2017-04-15:
That does work! thx Question. i changed Motordirection of my motors, cause i don't like it when gras and water spots are on my cam lense. then

yaw_motor_direction 1

is right?

Arakon, 2017-04-15:
+1 is standard (props spinning inward to front and back).
-1 is reversed (props spinning away from the front and back).

unitware, 2017-04-19:
@MimaRaza are you satisfied with the solution? can you close the issue in that case?

MimaRaza, 2017-04-19:
THX, solved it!

closed 2017-04-19 by MimaRaza