betaflight/betaflight #2884: Problems connecting Revo BTFL 3.1.7 with Multiplex SRXL RX5 m-link

Juan-ma, 2017-04-14:
Hello,

I have a Revo board with BTFL 3.1.7 and a Multiplex RX5 M-Link configured as SRXL receiver. I have configured Port 2 (UART1) as "Serial Receiver" and in the Configuration section, Receiver as "Serial-based receiver". In the "Serial receiver provider" I have tryed both SRXL and XBUS_MODE_B with no success. With XBUS_MODE_B provider selected, I see all chanels fixed at 1500 in the receiver section. With SRXL provider is selected, when I move right stick all channels move randomly.
I have tryed with older versions of betaflight even with cleanflight but the same thing happens. I have sniffed the serial output of the receiver and it's OK accord the Multiplex description of the protocol. I have tryed with other serial ports of the Revo with the same result.
I have had a look to the code and it seems that I should select "XBUS_MODE_B" for the Multiplex receiver and SRXL seems to be used with Spektrum receivers:

https://github.com/betaflight/betaflight/blob/master/src/main/rx/xbus.c#L62
https://github.com/betaflight/betaflight/blob/master/src/main/rx/rx.c#L219

Maybe the problem is the baudrate of the serial port in the revo (multiplex receiver is 115200).

AndersHoglund, 2017-04-15:
When using SRXL you must connect the Sat to UART Tx line, NOT the usual Rx line. 
Works fine for me on a Revo Acro and BF3.1.7

Juan-ma, 2017-04-15:
Thanks,

I have tryed both tx and rx. In SRXL mode, If i connect the signal pin of the receiver to the third pin of serial bus of the REVO all channels are stuck at 1500 and if connect it to the fourth pin all channels become erratic.

Are you using a Multiplex receiver? I think i should use XBUS_MODE_B mode with Multiplex receivers  but I'm not sure.  In this mode all channels are stuck to 1500 too.

AndersHoglund, 2017-04-15:
I did not realize it was about a Multiple Rx. Sorry. I am using Spektrum SPM4649T in SRXL mode, for telemetry. Can't help you with Mpx.
Also, is there a naming conflict here? SRXL used by both Mpx and Spktm, and those two being different things?? Then BFC should use a more exact name "Spektrum SXRL" not just "SXRL" in the "Serial Receiver Provider" drop down menu.

Juan-ma, 2017-04-15:
I'm not sure but it seems that SRXL is a generic protocol. The main difference is the header, but the rest should be very similar or even identic.

Anyway, thanks for your help.

AndersHoglund, 2017-04-16:
There is an old issue in CF, did you read that? https://github.com/cleanflight/cleanflight/issues/681 
Seems as some people got it working by reconfiguring both Tx and Rx:

- Enable serial output in my receiver with the multimat programmer (also possible with a pc if I'm correct)
- Set FastResponse to OFF in the modulation menu of my transmitter (Multiplex Royal Pro 9)

Juan-ma, 2017-04-16:
Thanks, I read that. I did both steps but the same thing happens.

Juan-ma, 2017-04-16:
Ok, I have succesfully connected the RX and the REVO board with XBUS_MODE_B provider. I dont know what I did wrong when I tryed it before because I tryed it several times. Anyway, in order to use the serial servo output with a Multiplex receiver, you have to:

- Enable the serial output of the receiver
- Set the provider XBUS_MODE_B in the betaflight configuration (not SRXL, that is used for Spektrum receivers. This is the most confusing point)
- Set FastResponse OFF in the modulation menu of trasnsmitter
- RX pin of the REVO serial bus conected to signal pin of the B/D port of the receiver (Also GND and Vcc)

Thank you very much for your help.

Juan-ma, 2017-04-18:
Ok, I have reproduced the problem. In the port configuration of the REVO you have 4 serial ports: USB VCP, UART1, UART3 and UART6. I can only use the 'XBUS_MODE_B' mode with UART3. I mean:

UART1 + SRXL mode (not the correct mode for Multiplex) = erratic values in all the channels
UART1 + XBUS_MODE_B mode (the correct mode for Multiplex) = fixed values. Throtle at minimum and the rest at 1500
UART3 + SRXL mode (not the correct mode for Multiplex) = erratic values in all the channels
UART3 + XBUS_MODE_B mode (the correct mode for Multiplex) = It works fine!

So the problem is I cant use the XBUS_MODE_B mode in UART1, I dont know why.

AndersHoglund, 2017-04-18:
UART1 has an sbus inverter that will mess up things, if used incorrectly. Should be controllable by setting sbus_inversion = ON or OFF. If UART3 works then UART6 should too. And the "SRXL" selection seems to mean "Spektrum Bidirectional SRXL" only. Not to be used with Mpx gear.

Juan-ma, 2017-04-19:
Is a bit strange:
UART1 dont work in XBUS_MODE_B regardless of setting sbus_inversion = ON or OFF.
UART3 do work in XBUS_MODE_B regardless of setting sbus_inversion = ON or OFF.

AndersHoglund, 2017-04-20:
I can run Spektrum satellites on both UART1 and UART3. UART6 not tested.
The sbus_inversion parameter seems to be valid only when provider SBUS is selected, ON or OFF has no effect on Spektrum sat either.  Strange, Mpx XBUS_MODE_B should work equally well on UART1 too, can not see anything in the code that would prevent that. I have no Mpx gear, can not verify nor debug this case.
Are you absolutely sure you don't have something else enabled on UART1 ? Like MSP etc etc....

Juan-ma, 2017-04-20:
This is my config:

[image]
[image]
[image]

catdog79, 2017-06-17:
Hallo together.
I have problems to get a MPX RX4/9 Flexx to work on a CC3D
I use Betaflight 3.1.7 and had same issues like in this thread above.

So i tried to activate UART3 and  XBUS_MODE_B.
But after reconnect the CC3D there is in the configuration tab no receiver choosen.
It seems i cant save the option "serial based receiver" and " XBUS_MODE_B".
If i choose "serial based receiver" and another combination like "srxl" or other i can save it.

Anyone an idea?

mikeller, 2017-06-17:
@catdog79: XBUS support had to be removed from CC3D to make the firmware fit into the small amount of flash space on the MCU. You could possibly build your own version of the firmware and make it fit by removing other features, or you could upgrade to a board with an F3 or even F4 MCU.

catdog79, 2017-06-18:
Ok thank u for the info.
I think i upgrade to an F3 Racing Acro Flight Controller SP RacingF3.

al21883, 2017-10-25:
@catdog79 I'm happy to tell yo that Betaflight 3.17  CC3D supports " XBUS_MODE_B" with Multiplex M-link receivers.
It may be some kind of bug, because i havd to do several stranger things to make it work. 
-I had to activate "serial based receiver" on Cleanflight interface, and save changes. 
Then on betaflight interface:
-I activated (and saved) sbus_inversion and then desactivated (and saved) (Don't know if this step is necessary) 
-Then after testing XBUS_MODE_B (with no success) I tested SBUS and SRXL and got erratic signals on SRXL mode.
-Finally got back to XBUS_MODE_B and It started to work perdecly. 
Some of the previous steps may enable the XBUS_MODE_B to work, but i don't know what. 
All the steps were done with the RX connected to the UART3 (Flexi Port)  configured as Serial RX, on a HK clone CC3D.

closed 2018-04-21 by DieHertz