betaflight/betaflight #2885: Helper-functionality to solve arming issues

jonerFPV, 2017-04-15:
Maybe a CLI command or even feature to configurator which would tell what problems have been found at the copter, for example:

- throttle too high to arm
- attitude angle too much to arm
- no RC connection
- looptime warning (CPU overloaded)
- gyro error or other HW info

That would help people to understand what is wrong and it would be easy to point newbies to start with ("did you check problems output? to to CLI and type problems")

I could maybe implement this if it's a good idea and there's no-one else willing to do it.

McGiverGim, 2017-04-20:
Another solution, I don't know if better or worse, is write to the Blackbox the info with all errors found. Can be into the actual blackbox file or simply a new one with only the errors.

jonerFPV, 2017-04-20:
Blackbox would be a bonus - however not all have blackbox cabability, especially beginners whom this feature would benefit the most.

I think the correct way would be to implement visualization of this to configurator.

mikeller, 2017-05-22:
My personal favourites would be these two (possibly in combination with the abovementioned):
- beep / led codes (works out of the box with any FC);
- OSD display.

davecraige, 2017-11-14:
Here is a visualization of how Automatic.com does it.  They keep you informed if there is any issues with your car.

davecraige, 2017-11-14:
Here is how Automatic.com does the design of this.  They make it clear if there are any issues with your car.  

[image]

closed 2018-06-10 by mikeller