betaflight/betaflight #2886: softserial not working on SPRACINGF3EVO

Turismo05, 2017-04-15:
Hi, according to the fix list for 3.17 soft serial is working on this target but its not listed in the ports page when enabled.

Any ideas???

AndersHoglund, 2017-04-15:
No room for softserial in the flash, it had to be disabled before release. There might have been a fix for a softserial problem prior to that. A bit misleading change note, I agree. Not incorrect, only incomplete. Half the story only. Fix it (and then disable it).

Turismo05, 2017-04-16:
Thats a shame, how much flash does it need? could I remove something else and recompile?

AndersHoglund, 2017-04-16:
Yes, of course you can. GPS is a common candidate for removal on a few targets, do it in the same way and re-enable soft serial:

    $ grep -r "undef GPS" src/main/target/
    src/main/target/ALIENFLIGHTF1/target.h:#undef GPS
    src/main/target/OMNIBUS/target.h:#undef GPS
    src/main/target/CC3D/target.h:#undef GPS
    src/main/target/MICROSCISKY/target.h:#undef GPS
    src/main/target/common.h:#undef GPS
    src/main/target/AIR32/target.h:#undef GPS
    src/main/target/SPRACINGF3NEO/target.h:#undef GPS
    src/main/target/NAZE/target.h:#undef GPS
    src/main/target/MOTOLAB/target.h:#undef GPS

Turismo05, 2017-04-16:
cool thanks

closed 2018-04-21 by DieHertz