betaflight/betaflight #2887: BF 3.1.7 and OmnibusF4SD acc trim board reset

alfag4, 2017-04-15:
Hello, trimming acc via stick command reset the FC just after moving right stick without save the new settings.

jflyper, 2017-04-17:
It doesn't reproduce with OMNIBUS F4 V2 (OMNIBUSF4SD target).

Please specify the procedure to reproduce.

alfag4, 2017-04-18:
In disarm state, left stick up, when moving right stick up/down or left/right i hear the trim beep but still after the board reset and the ESCs beeping. The new settings are saved.

jflyper, 2017-04-18:
@alfag4 You mean

(1) left stick up
(2) right stick any of up/down/left/right
(3) trim beep sounds
(4) board resets, ESC plays initial 3+2 tones

Do all of up/down/left/right trigger this with single trimming action?

alfag4, 2017-04-19:
@jflyper  Yes
The board resets every time moving the right stick. Now i'm moving to INAV to see if there is the same issue.

jflyper, 2017-04-26:
@alfag4 How was iNav?

alfag4, 2017-04-26:
With INAV i'm able to trimming without any issue i hear only the trim beep when moving the right stick and the ESCS does not resets. I remember that the same problem was presents in the 3.1.6 version.

jflyper, 2017-04-27:
@alfag4 Can you give us CLI diff result? I will load the config onto my board to see if it reproduces.

alfag4, 2017-04-29:
My quad:
Falcon 250 Pro modified
FC: Omnibus F4 SD
ESC: LittleBee_S 30A

# name
name F250

# resources

# mixer
mixer CUSTOM

mmix reset

mmix 0  1.000 -1.000  0.772 -1.000
mmix 1  1.000 -1.000 -0.772  1.000
mmix 2  1.000  1.000  0.772  1.000
mmix 3  1.000  1.000 -0.772 -1.000

# servo

# servo mix

# feature
feature GPS
feature CURRENT_METER

# beeper
beeper -SYSTEM_INIT
beeper -ON_USB

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1900 2100
aux 1 1 1 900 1100
aux 2 2 1 1400 1600
aux 3 3 3 1900 2100
aux 4 3 2 1900 2100
aux 5 10 3 1900 2100
aux 6 11 2 1900 2100

# adjrange

# rxrange

# rxfail
rxfail 4 s 875

# master
set min_check = 1080
set max_check = 1990
set debug_mode = NOTCH
set min_throttle = 1030
set max_throttle = 1950
set motor_pwm_protocol = MULTISHOT
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set serialrx_provider = IBUS
set gyro_notch1_hz = 390
set gyro_notch1_cut = 309
set gyro_notch2_hz = 209
set gyro_notch2_cut = 109
set deadband = 4
set yaw_deadband = 4
set failsafe_off_delay = 200
set acc_trim_roll = -6
set baro_hardware = AUTO
set mag_hardware = AUTO
set mag_declination = 2300
set pid_process_denom = 1
set blackbox_rate_denom = 4
set magzero_x = 9
set magzero_y = 12
set magzero_z = -29
set osd_units = METRIC
set osd_crosshairs = 2048
set osd_current_pos = 2145
set osd_craft_name_pos = 2440
set osd_altitude_pos = 2209
set vcd_video_system = 1

# profile
profile 0

set d_notch_hz = 209
set d_notch_cut = 130
set p_pitch = 63
set i_pitch = 52
set d_pitch = 37
set p_roll = 48
set p_yaw = 77
set i_yaw = 43
set p_level = 45
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 57
set rc_rate_yaw = 65
set rc_expo = 20
set rc_yaw_expo = 20
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
set tpa_rate = 0

jflyper, 2017-04-29:
@alfag4 
It doesn't reproduce.

- I don't have an IBUS receiver, so tested with SBUS.
- I've noticed that status LED doesn't flash for confirmation.

Can you revert to default state, configure the receiver only and see if it reproduces?

alfag4, 2017-06-08:
Issue resolved with the latest 3.2.0

closed 2017-06-08 by alfag4